cartographer-project/cartographer #1400: Visualize on rviz with a 3d camera (ROS)

Semih67, 2018-08-28:
I have a ifm3d camera and I want to use the pointclouds of the camera to have a map on rviz. I didn't do any configurations in the cartographer folder. And I also don't know how to launch the google cartographer to visualize it on rviz and using the output of the camera. I only could find demos that I have to download and launch them. But I have my own camera and simultaneously my own data. I tried the command roslaunch cartographer_rviz but it's not working. So can someone explain me how to use the google cartographer?

kdaun, 2018-08-31:
See the [Cartographer documentation](https://google-cartographer.readthedocs.io/en/latest/index.html) and the [Cartographer ROS](https://google-cartographer-ros.readthedocs.io/en/latest/index.html) documentation for an overview over the Cartographer setup. If you don't feel comfortable with ROS and RViz, start with the [ROS tutorials](http://wiki.ros.org/ROS/Tutorials).
So far, we were not successful to create accurate maps with Cartographer using RGBD cameras. However, we are excited to hear about your results.

Semih67, 2018-09-03:
Do you have an example how to use google cartographer? Do I have to write a code to feed the the cartographer or just do some configurations?

Semih67, 2018-09-03:
Which inputs are required for google cartographer? Do I only need pointcloud2 and the depth? Which data is also needed to feed the cartographer?

Semih67, 2018-09-03:
And do I feed the cartographer with the sensor data? In which file do I have to config it?

Semih67, 2018-09-03:
And how do I feed the cartographer with the sensor data? In which file do I have to config it?

Entropy512, 2018-10-12:
If you want to do 3D, you will need an IMU.

At least my experience with IFM cameras is that the noise profile is such that you're unlikely to get a decent map (if at all) with Cartographer, similar with the issues some have had with RGBD cameras.

I have some IFM O3D units where I work, I've never even tried to use them for mapping, partly due to the fact that spinning multiplanar LIDAR units have much better range, field of view, and refresh rate.

Pushing an O3D to maximum range involves significant framerate compromises.  Implementing high dynamic range mode compromises framerate further.  In this operation mode, the camera gets VERY hot and unless IFM has redesigned their heatsink in the past two years, their official heatsink solution is extremely poor.

Every TOF camera I've worked with has had a tendency to have really nasty artifacts in the data near sharp edges, and also frequently serious problems with range aliasing.

closed 2020-08-26 by wohe