cartographer-project/cartographer #1418: Cartographer not generating submap

daiencan, 2018-09-11:
Hi, I am a master's student at the University of Cqupt and I am trying to implement 3D SLAM using the RSLiDAR-16. The cartographer node subscribes to the /rslidar_points topic , /imu topic and /odom topic. however, the sub-map does not build. There is no error or warning in the terminal. I will really appreciate some help as to why the sub-map is not being constructed. Thank you very much

daiencan, 2018-09-11:
[ INFO] [1536646053.579129226]: I0911 14:07:33.000000 32392 configuration_file_resolver.cc:41] Found '/home/dec/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1536646053.579166344]: I0911 14:07:33.000000 32392 configuration_file_resolver.cc:41] Found '/home/dec/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1536646053.587980171]: I0911 14:07:33.000000 32392 submap_3d.cc:321] Added submap 1
[ INFO] [1536646053.588170012]: I0911 14:07:33.000000 32392 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1536646065.124992437, 1536588712.090116382]: I0911 14:07:45.000000 32392 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636717472161880590'.
[ INFO] [1536646065.131718862, 1536588712.100208761]: I0911 14:07:45.000000 32392 local_trajectory_builder_3d.cc:254] Extrapolator not yet initialized.
[ INFO] [1536646068.647109090, 1536588715.613546245]: I0911 14:07:48.000000 32392 collated_trajectory_builder.cc:72] imu rate: 49.99 Hz 2.00e-02 s +/- 1.07e-03 s (pulsed at 100.70% real time)
[ INFO] [1536646068.647189888, 1536588715.613546245]: I0911 14:07:48.000000 32392 collated_trajectory_builder.cc:72] odom rate: 49.86 Hz 2.01e-02 s +/- 6.07e-03 s (pulsed at 98.14% real time)
[ INFO] [1536646068.647227557, 1536588715.613546245]: I0911 14:07:48.000000 32392 collated_trajectory_builder.cc:72] points2 rate: 9.92 Hz 1.01e-01 s +/- 2.10e-04 s (pulsed at 100.30% real time)

daiencan, 2018-09-11:
@gaschler Above is information after I rosrun the cartographer node and robag. And the position of rslidar is changed so quickly. the value of rslidar position is very large.

gaschler, 2018-11-30:
Please follow the issue template to verify with rosbag_validate that your input data is fine.
Then, look out for warnings in the terminal output about topic mismatches.