cartographer-project/cartographer #1419: Reproduce this error by running

sweetquiet, 2018-09-11:
Source space: /home/youibot/catkin_ws1/src
Build space: /home/youibot/catkin_ws1/build_isolated
Devel space: /home/youibot/catkin_ws1/devel_isolated
Install space: /home/youibot/catkin_ws1/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'cartographer_ros_msgs'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
ninja: no work to do.
==> ninja -j4 -l4 in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
[13/13] cd /home/youibot/catkin_ws1/bu...ographer_ros_msgs/msg/SensorTopics.msg
==> ninja install in '/home/youibot/catkin_ws1/build_isolated/cartographer_ros_msgs'
[14/14] Install the project...
-- Install configuration: ""
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/_setup_util.py
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/env.sh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.bash
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.sh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/setup.zsh
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/.rosinstall
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/LandmarkEntry.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/LandmarkList.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/StatusCode.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/StatusResponse.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapList.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapEntry.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SubmapTexture.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/SensorTopics.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/msg/TrajectoryOptions.msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/SubmapQuery.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/FinishTrajectory.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/StartTrajectory.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/srv/WriteState.srv
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/cartographer_ros_msgs/cmake/cartographer_ros_msgs-msg-paths.cmake
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteState.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StatusResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/LandmarkEntry.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapEntry.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQueryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectoryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectory.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQuery.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapTexture.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SensorTopics.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectory.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectoryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapQueryRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/FinishTrajectoryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteStateRequest.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/LandmarkList.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StatusCode.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/WriteStateResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/StartTrajectoryResponse.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/SubmapList.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/include/cartographer_ros_msgs/TrajectoryOptions.h
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/manifest.l
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/msg
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/msg/StatusCode.l
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/msg/SensorTopics.l
-- Up-to-date: /home/youibot/catkin_ws1/install_isolated/share/roseus/ros/cartographer_ros_msgs/msg/SubmapEntry.l
[... 452 lines cut ...]
transform_interpolation_buffer_test.cc:(.text._ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_[_ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_]+0xc94)：对‘testing::internal::IsTrue(bool)’未定义的引用
transform_interpolation_buffer_test.cc:(.text._ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_[_ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_]+0xfc9)：对‘testing::internal::IsTrue(bool)’未定义的引用
transform_interpolation_buffer_test.cc:(.text._ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_[_ZNK7testing8internal29PredicateFormatterFromMatcherIN12cartographer9transform17IsNearlyMatcherP2INS3_6Rigid3IdEEdEEEclIS6_EENS_15AssertionResultEPKcRKT_]+0x115a)：对‘testing::internal::GTestLog::~GTestLog()’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0x10)：对‘typeinfo for testing::Test’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0x40)：对‘typeinfo for testing::Test’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0x70)：对‘typeinfo for testing::Test’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0xb0)：对‘testing::Test::SetUp()’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0xb8)：对‘testing::Test::TearDown()’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0xf0)：对‘testing::Test::SetUp()’未定义的引用
CMakeFiles/cartographer.transform.transform_interpolation_buffer_test.dir/cartographer/transform/transform_interpolation_buffer_test.cc.o:(.data.rel.ro+0xf8)：对‘testing::Test::TearDown()’未定义的引用
[... cut ...]
gmock_main.cc:(.text._Z13RUN_ALL_TESTSv[_Z13RUN_ALL_TESTSv]+0xd)：对‘testing::UnitTest::Run()’未定义的引用
collect2: error: ld returned 1 exit status
[4/142] Building CXX object CMakeFiles...er/transform/rigid_transform_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/youibot/catkin_ws1/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/youibot/catkin_ws1/build_isolated/cartographer && /home/youibot/catkin_ws1/install_isolated/env.sh ninja -j4 -l4

gaschler, 2018-11-30:
Closing as obsolete

closed 2018-11-30 by gaschler

xijunke, 2019-04-12:
# Comment by XJK：
Comment the following lines for CMakeLists.txt from "/home/q/catkin_ws/src/cartographer"
###################################################################################
# Comment by KXJ
# foreach(ABS_FIL ${ALL_TESTS})
#   file(RELATIVE_PATH REL_FIL ${PROJECT_SOURCE_DIR} ${ABS_FIL})
#   get_filename_component(DIR ${REL_FIL} DIRECTORY)
#   get_filename_component(FIL_WE ${REL_FIL} NAME_WE)
#  # Replace slashes as required for CMP0037.
#   string(REPLACE "/" "." TEST_TARGET_NAME "${DIR}/${FIL_WE}")
#   google_test("${TEST_TARGET_NAME}" ${ABS_FIL})
#   if(${BUILD_GRPC})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC grpc++)
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC async_grpc)
#   endif()
#   if(${BUILD_PROMETHEUS})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC prometheus-cpp)
#   endif()
#   target_link_libraries("${TEST_TARGET_NAME}" PUBLIC ${TEST_LIB})
# endforeach()
###################################################################################

xijunke, 2019-04-12:
# Comment by XJK ：
**Comment** the following lines for CMakeLists.txt from "/home/q/catkin_ws/src/cartographer"
###################################################################################
# foreach(ABS_FIL ${ALL_TESTS})
#   file(RELATIVE_PATH REL_FIL ${PROJECT_SOURCE_DIR} ${ABS_FIL})
#   get_filename_component(DIR ${REL_FIL} DIRECTORY)
#   get_filename_component(FIL_WE ${REL_FIL} NAME_WE)
#  # Replace slashes as required for CMP0037.
#   string(REPLACE "/" "." TEST_TARGET_NAME "${DIR}/${FIL_WE}")
#   google_test("${TEST_TARGET_NAME}" ${ABS_FIL})
#   if(${BUILD_GRPC})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC grpc++)
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC async_grpc)
#   endif()
#   if(${BUILD_PROMETHEUS})
#     target_link_libraries("${TEST_TARGET_NAME}" PUBLIC prometheus-cpp)
#   endif()
#   target_link_libraries("${TEST_TARGET_NAME}" PUBLIC ${TEST_LIB})
# endforeach()
###################################################################################