cartographer-project/cartographer #1422: Allow snapshotting of large pose graphs from cartographer server

cschuet, 2018-09-12:
Currently when snapshotting the cartographer server state using MapBuilderService::WriteState, we stream back a sequence of SerializedData protos. One of them contains the PoseGraphProto which can easily grow beyoned the maximum supported protobuf message size of gRPC (currently 10 MB).

I would be better to add a WriteStateToFile() gRPC method which allows the client to ask the server to directly write the pbstream to a local file.

This method should also be invoked from the cartographer_grpc_node when the ROS service for saving state is called.

cschuet, 2018-09-12:
@MichaelGrupp FYI

MichaelGrupp, 2018-09-21:
Resolved, you can close this.

cschuet, 2018-09-23:
Resolved by https://github.com/googlecartographer/cartographer/pull/1424.

closed 2018-09-23 by cschuet