cartographer-project/cartographer #1432: Local SLAM quality issues, z-ghosting of objects in adjacent submaps

ojura, 2018-09-21:
Hello Cartographer team,

I've stumbled into a wall while trying to find a tuning for 3D SLAM that will produce a satisfactory point cloud. The point clouds are a bit too fuzzy in general (e.g. ~20 cm wall and road thickness) and the details are often too smeared. One particular issue I have noticed is Z-ghosting of objects, usually when the vehicle makes a turn. The IMU should be fairly well calibrated to the lidar. 

The dataset is available here:
https://ferhr-my.sharepoint.com/:f:/g/personal/jorsulic_fer_hr/EpPrJrsM3pxDpNGn22fDSZYBBjiNbdx4l7I7f5ProfLPHg?e=UQMA4j

For evaluating this, I disabled building constraints, and took a 100s long slice of the dataset where the vehicle makes a left turn.  The dataset is `type1_traderjoes_first100.bag`, used only the VLP-16 lidar, and the tuning is `type1_no_loop_closure.lua`. To run:

```
rosrun cartographer_ros cartographer_offline_node -configuration_directory . -configuration_basenames type1_no_loop_closure.lua -bag_filenames type1_traderjoes_first100.bag points2:=points2_1_VLP16 -use_bag_transforms=true
```

Here is a screenshot of the resulting point cloud. The points are colored by time.
[image]

Here is a visualization of the submaps, where each submap is colored with a different color:
[image]

This is definitely too much drift going on between adjacent submaps. Even if the whole trajectory has Z-drift, the adjacent submaps should not look broken.

I'd appreciate your help in making this look pretty, if you have the time to spare :)

gaschler, 2018-09-24:
I assume you didn't see any alignment issues in rviz (as far as the projected submaps allow you to tell)?

If you use the point cloud asset writer, I guess you are one of the first users to provide per-point timestamps in 3D through cartographer_ros, so it's worth questioning the conversion there.

ojura, 2018-09-24:
The points you see here have been gathered from `LocalSlamResult`s which contained `InsertionResult`s - this should be equivalent to what the asset writer is doing, as I am using the optimized poses of the respective trajectory nodes. Since there's no loop closure here, the only effect of optimization should be smoothing it a bit and aligning gravity, according to the IMU acceleration/angular velocity cost functions. Even if optimization is completely disabled, the local SLAM quality issues persist.

You can not easily see these z ghosting issues in rviz, since the submaps are 2D projections from above. 
Also, not using the provided per-point timestamps does not make much of a difference. If they were problematic, we would probably see problems in the direction of car's velocity.

gaschler, 2018-09-24:
Did you try reducing TRAJECTORY_BUILDER_3D.ceres_scan_matcher.translation_weight 
and POSE_GRAPH.optimization_problem.acceleration_weight?

ojura, 2018-09-24:
Yes, and I have tried increasing the occupied space weights. No matter what I try, I just really can't get local SLAM to produce a good result without messing up the finer details such as lamp poles or traffic lights.

ludennis, 2018-10-22:
@ojura, I am experiencing the same issue, and haven't gotten a solution to it. Were you able to find out how to correct the z misalignment?