cartographer-project/cartographer #1434: I can not install ros-indigo-tf2-eigen even though I have wasted a lot of time in it!

SnowCarter, 2018-09-29:
[image]
Just like this. I do not know how to solve it...

MichaelGrupp, 2018-10-01:
Do you have ROS installed? If yes, is it version "Indigo"? (run `rosversion -d`)

`apt` seems to have trouble finding the package in the configured software repositories.

gaschler, 2018-11-30:
Closing as not actionable

closed 2018-11-30 by gaschler

vishalsleotechsa, 2019-08-10:
what is the way to solve this issue 
Can't install on ROS Indigo: ros-indigo-tf2-eigen not found
ubuntu 14.04  64-bit
16 GB RAM
gcc version 4.8.4
ROS -indigo distribution

I have attached the terminal text,
"
$rosdep install --from-paths src --ignore-src --rosdistro=indigo -y
executing command [sudo -H apt-get install -y ros-indigo-tf2-eigen]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  ros-indigo-tf2-eigen
0 upgraded, 1 newly installed, 0 to remove and 317 not upgraded.
Need to get 8,234 B of archives.
After this operation, 98.3 kB of additional disk space will be used.
Err http://packages.ros.org/ros/ubuntu/ trusty/main ros-indigo-tf2-eigen amd64 0.5.20-0trusty-20181117-044151-0800
  404  Not Found [IP: 140.211.166.134 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-indigo-tf2-eigen/ros-indigo-tf2-eigen_0.5.20-0trusty-20181117-044151-0800_amd64.deb  404  Not Found [IP: 140.211.166.134 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-indigo-tf2-eigen] failed
"