cartographer-project/cartographer #1441: Jump at similarity position & suddenly jump ~40cm

zhengshunkai, 2018-10-07:
**Hi,I am using the version around December 2017.Pure localization model(2d_Laser_R2000+IMU_LPMS).
The problem is this:**
### 1.Jump at similarity position
As shown in the following figure，when a robot stays at A position for a period of time, its position will jump to B.
[image]
**Can we solve the problem by tuning parameters ？**
### 2.After running for a period of time, suddenly jump ~40cm
As shown in the following figure，the jump of ~348mm occurs in the 5524th data.
[image]
I think there are two possible reasons，but I'm not sure.：
**a：the submap broken 
b：There is a drift when matching ，we can see that the map has moved a little when matching. ** 
because the map building is not perfect ？or the parameters  setting is not reasonable？

this is my parameters:
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching =false
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1  
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 0.1
SPARSE_POSE_GRAPH.optimize_every_n_scans = 10 
SPARSE_POSE_GRAPH.global_sampling_ratio = 0.001
SPARSE_POSE_GRAPH.constraint_builder.sampling_ratio = 0.1
MAP_BUILDER.num_background_threads = 4 
SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.55 
SPARSE_POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7 
SPARSE_POSE_GRAPH.global_constraint_search_after_n_seconds = 10
SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2 
TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.max_length = 0.5
TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.min_num_points = 200 
TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.max_range = 50  
TRAJECTORY_BUILDER_2D.voxel_filter_size = 0.025 
TRAJECTORY_BUILDER_2D.submaps.resolution = 0.05 
TRAJECTORY_BUILDER_2D.submaps.num_range_data= 90 
TRAJECTORY_BUILDER_2D.min_range = 0
TRAJECTORY_BUILDER_2D.max_range = 15
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1 

thank you for your help~

zhengshunkai, 2018-10-09:
help~

Entropy512, 2018-10-12:
Looking at the map, the LIDAR's view from A is going to be so similar to B that it may be very difficult to distinguish these - however with an IMU it probably shouldn't be jumping?  Perhaps adding vehicle odometry might help?

zhengshunkai, 2018-10-13:
> Looking at the map, the LIDAR's view from A is going to be so similar to B that it may be very difficult to distinguish these - however with an IMU it probably shouldn't be jumping? Perhaps adding vehicle odometry might help?

Thank you for your reply. when the location is lost, you need to relocate, in which case it will jump to a higher similarity position, which should be a normal phenomenon? Is it useful to adjust IMU related weights?

sunbingfeng, 2020-11-02:
It seems that the two places are similar, and is hard to distinguish even for human being.

Adding odometry might help, but tuning the weights of odom is troublesome.

So how did you solve this problem finnaly? @zhengshunkai

charon-cheung, 2021-01-06:
> It seems that the two places are similar, and is hard to distinguish even for human being.
> 
> Adding odometry might help, but tuning the weights of odom is troublesome.
> 
> So how did you solve this problem finnaly? @zhengshunkai

我也遇到了相似的问题，你解决了吗