cartographer-project/cartographer #1444: Windows pipeline

ojura, 2018-10-10:
What do you think about adding an Appveyor or Google Compute Engine Windows build to Cartographer's CI, using Microsoft's ROS1 port?

MichaelGrupp, 2018-10-16:
I don't see a problem with trying that out, but I'm not sure about the CI part:

* how stable/mature is the ROS Windows port?
* what if a build breaks _only_ on Windows - will it block PRs? I (and probably most other Cartographer contributors too) don't have a Windows dev environment to test and fix locally

MichaelGrupp, 2018-10-16:
Ok, my suggestion would be to have only builds from master built in the Windows pipeline for informational purposes as you said (post-commit, not pre-commit ;) ).

ojura, 2018-10-16:
Why would you not build PRs? If you are inclined to fix your PRs for windows (if necessary) before they are merged, you can do so using Windows CI logs, even if you don’t have Windows. If you’re not, well, you could just ignore the Windows build status.

MichaelGrupp, 2018-10-16:
I'm also fine with that.

ooeygui, 2018-10-19:
@MichaelGrupp (This is Lou from Microsoft, I've been working with @ojura on the Cartographer port)
The ROS1 on Windows is experimental as we are still in bring-up. We're hoping to finish core ROS1 and certain slices through manipulation and mobility by the end of the year.

FYI: On our backlog is the ability to easily set up ROS project CI builds using Azure DevOps - which is now free for [OpenSource projects](https://azure.microsoft.com/en-us/pricing/details/devops/azure-pipelines/).

Many customers have asked for Cartographer on Windows. I'm happy to help make that happen. I think it would be a good test case of the Windows/ROS CI build tooling.

Thank you,
Lou

MichaelGrupp, 2018-10-22:
Sounds like something that could be discussed with the maintainers in the next ["open house" session](https://github.com/googlecartographer/cartographer#open-house).