cartographer-project/cartographer #1449: Timestamps are zero when parsing posegraph saved to .pbstream file

doge-of-the-day, 2018-10-20:
Hi guys, 
I am trying to parse pose graph trajectories and I encountered an issue, that has not existed some months ago. Though I am using the offline_tool from the cartographer_ros package, the issue is more probably related to the cartographer core code. The same launch and configuration files, which worked some months ago, do not anymore. 
To come to the core of my problem, I am parsing .pbstream files and retrieving the pose graphs and their trajectories. The problem is, that all timestamps of trajectory nodes are zero, which was not the case before. 
So either I am using the API wrong or the timestamps are lost serializing the trajectory nodes. 
Would be nice, if you could have a look at this or tell me what the hack I am doing wrong.

The code I am using:
```
    std::shared_ptr<cartographer::mapping::proto::PoseGraph> p(new cartographer::mapping::proto::PoseGraph);
        cartographer::io::ProtoStreamReader reader(path.string());
    while(reader.ReadProto(p.get()) && !reader.eof()) {
        for(auto &t : p->trajectory()) {
            ros::Time ros_last_time;
            tf::Transform ros_last_tf;

            for(int i = 0 ; i < t.node_size() ; ++i) {
                auto time = cartographer::common::FromUniversal(t.node(i).timestamp());
                auto translation = t.node(i).pose().translation();
                auto rotation    = t.node(i).pose().rotation();

                ros::Time      ros_time = cartographer_ros::ToRos(time);
                tf::Quaternion ros_rotation(rotation.x(), rotation.y(), rotation.z(), rotation.w());
                ros_rotation.normalize();
                tf::Vector3    ros_translation(translation.x(), translation.y(), translation.z());
                tf::Transform  ros_tf(ros_rotation, ros_translation);

                trajectory_writer.write(ros_time.toSec(), ros_translation.x(), ros_translation.y(), tf::getYaw(ros_rotation));
                if(!ros_last_time.isZero()) {
                    tf::Transform  ros_rel_tf = ros_last_tf.inverse() * ros_tf;
                    relative_pose_writer.write(ros_last_time.toSec(), ros_time.toSec(),
                                               ros_rel_tf.getOrigin().x(), ros_rel_tf.getOrigin().y(), 0.0,
                                               0.0, 0.0, tf::getYaw(ros_rel_tf.getRotation()));
                }
                ros_last_time = ros_time;
                ros_last_tf   = ros_tf;

                std::cout << "[" << ros_time << "]: " << translation.x() << "," << translation.y() << "," << tf::getYaw(ros_rotation) << "\n";
            }
            size += t.node_size();
        }
        p.reset(new cartographer::mapping::proto::PoseGraph);
    }
```

doge-of-the-day, 2018-10-20:
A, I just found another issue, die translation values are also 0 and the rotation is nan...

cschuet, 2018-10-20:
We updated the pbstream file format some time ago. You might have to run the migration tool on your pbstreams to convert them to the new format.

doge-of-the-day, 2018-10-20:
The .pbstream files were just created recently, but I guess we may have an issue with the protobuf version we used to compile which is v3.4.1. I am currently investigating what happens if we use a newer version.

doge-of-the-day, 2018-10-20:
Ok, the problem seems to persist. I have compiled the newest version of cartographer from master branch, the same for protobuf and the ceres-solver. The problem remains and the timestamps are zero. 

Is there anything wrong with my code or is it possible, that the serialization does not work properly?
Can anybody confirm these problems??

doge-of-the-day, 2018-10-21:
Ok, I finally got it work. Had a look at your test cases and used the Deserializer. Now everything is fine. Thank you ;)

closed 2018-10-21 by doge-of-the-day