cartographer-project/cartographer #1455: Fixed 1st submap while GPS (fixed_pose_frame) exist?

qiao-tw, 2018-10-24:
Hi,
I met a problem here,
https://github.com/googlecartographer/cartographer/blob/ceddabbcbb05d5fcf229ddd24683c703dfd79ea0/cartographer/mapping/internal/optimization/optimization_problem_3d.cc#L298-L310

While I want to build the map with GPS data (i.e. fixed_pose_frame), I'll expect the 1st submap is not fixed. However, the logic here assuming 1st submap won't change at all.

Is my understanding correct? Any hints are welcome, thanks.

ojura, 2018-10-24:
You are correct, that is certainly possible. I am doing exactly that in my private fork (throwing away this hardcoded constraint of the 1st submap and letting Ceres figure out how to align the trajectory with GPS measurements). I was hoping to upstream this for the last couple of months, but unfortunately didn’t manage to find the time to prepare a high quality PR.

qiao-tw, 2018-10-25:
@ojura Thank you very much!
I've just tried removing
 `problem.SetParameterBlockConstant(C_submaps.at(submap_id_data.id).translation());`
It did solve this issue, but I've never tune a good parameters set to get high quaility map.

Would you please tell me what are the potential side-effect of this naive fix? Thanks.

ojura, 2018-10-26:
You need to constrain the trajectory to the world coordinate system by inserting additional constraints into the pose graph. The hardcoded submap zero at the origin is the only thing tying it down - if you remove it, the trajectory will float in the world uncostrained during optimization.

rgreid, 2019-01-22:
From the Ceres code example for pose graphs:
```
  // The pose graph optimization problem has six DOFs that are not fully
  // constrained. This is typically referred to as gauge freedom. You can apply
  // a rigid body transformation to all the nodes and the optimization problem
  // will still have the exact same cost. The Levenberg-Marquardt algorithm has
  // internal damping which mitigates this issue, but it is better to properly
  // constrain the gauge freedom. This can be done by setting one of the poses
  // as constant so the optimizer cannot change it.
```
https://github.com/ceres-solver/ceres-solver/blob/master/examples/slam/pose_graph_3d/pose_graph_3d.cc#L92

closed 2019-05-19 by qiao-tw