cartographer-project/cartographer #1464: how  to add odometry data?

plumewind, 2018-11-19:
Hello, friends
I use turtlebot2 in ROS gazebo,I have set the lua:

  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,

but , it always error:
F1119 10:05:45.271495 26453 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355991307850000 vs. 621355991307850000) 
[FATAL] [1542593145.271714611, 2330.794000000]: F1119 10:05:45.000000 26453 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355991307850000 vs. 621355991307850000)

Can someone help me? Thank you

gaschler, 2018-11-30:
Please post the results of rosbag_validate following the issue template.

qust001, 2019-04-16:
same to me

closed 2020-06-03 by wohe