cartographer-project/cartographer #1470: Add landmarks to existing map for pure localization mode

johuber, 2018-11-27:
In our application we have a known (mapped) environment and operate in pure localization mode.
Since we have poor localization performance in certain parts of the map and possibly also to accelerate initialization, we'd like to add distinct landmarks (QR-Tags or similar) at certain places.

Is there a way to define the landmark poses with respect to the map at this stage? We already have a pbstream of the map, but don't want to rebuild the map because it takes too much time considering that we will need to try different landmark configurations.

gaschler, 2018-11-30:
The current API does not allow editing existing pbstreams.
You could write this feature by adding backdated landmark observations but this would only be a hack.

johuber, 2018-11-30:
Ok, I thought so.

I also experimenting with running mapping mode again on the same map, but loading an existing .pbstream to start. In general this seems to work and the map is updated, but somehow the newly observed landmarks are not introduced to the pose graph, or at least the marker array for landmarks remains empty. 

What's the reason for this? Shouldn't they be added?

johuber, 2018-12-03:
I just found the answer to my question.

As @ojura stated in his [comment](https://github.com/googlecartographer/cartographer/pull/988#issuecomment-373159268) in PR #988 , I would now be interested in the case, where I can add landmarks on top of a frozen trajectory.

I will submit a PR for this. My proposal:
- always allow introduction of new landmarks
- set parameter blocks of a trajectory's landmarks constant whenever `freezeTrajectory` is called for it.

Any objections?

PR #1475 (2019-01-17): Allow introduction of new landmarks into loaded (frozen) map
In the current implementation, as soon as one trajectory is frozen it is not possible to use new landmarks anymore.
In this PR a suggest a change, that allows introducing new landmarks also in the presence of a frozen trajectory. This is done with the following steps:
- Store set of frozen landmarks
- Landmarks that are loaded from a pbstream, with the flag ` load_frozen_state=true` are inserted into the set of frozen landmarks via `setLandmarkPose` function
- frozen landmarks are set constant in the optimization problem
- new landma [... cut]
commit fd679757 (2019-01-17): Allow introduction of new landmarks into loaded (frozen) map (#1475)

In the current implementation, as soon as one trajectory is frozen it is not possible to use new landmarks anymore.
In this PR a suggest a change, that allows introducing new landmarks also in the presence of a frozen trajectory. This is done with the following steps:
- Store set of frozen landmarks
- Landmarks that are loaded from a pbstream, with the flag ` load_frozen_state=true` are inserted into the set of frozen landmarks via `setLandmarkPose` function
- frozen landmarks are set constant in the optimization problem
- n [... cut]
closed 2019-01-17 by wally-the-cartographer