cartographer-project/cartographer #1481: Build error, can not found protobuf which is already installed

jinfagang, 2018-12-15:
```
[138/261] Building CXX object CMakeFiles/cartographer.mapping.intern...artographer/mapping/internal/trajectory_connectivity_state_test.cc.o
FAILED: cartographer_autogenerate_ground_truth 
: && /usr/bin/c++  -O3 -DNDEBUG  -rdynamic CMakeFiles/cartographer_autogenerate_ground_truth.dir/cartographer/ground_truth/autogenerate_ground_truth_main.cc.o  -o cartographer_autogenerate_ground_truth  -Wl,-rpath,/usr/local/lib: libcartographer.a /usr/local/lib/libceres.a -lglog /usr/lib/x86_64-linux-gnu/libgflags.so.2.2.1 -lspqr -ltbbmalloc -ltbb -lcholmod -lccolamd -lcamd -lcolamd -lamd -lopenblas -lsuitesparseconfig -lrt -lcxsparse -lopenblas -lsuitesparseconfig -lrt -lcxsparse -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog /usr/lib/x86_64-linux-gnu/libgflags.so.2.2.1 -lpthread -lcairo /usr/local/lib/libprotobuf.so -lpthread && :
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::GetMetadata() const':
relations.pb.cc:(.text+0x81): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
relations.pb.cc:(.text+0x133): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::GroundTruth::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
relations.pb.cc:(.text+0x1eb): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::(anonymous namespace)::protobuf_RegisterTypes(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
relations.pb.cc:(.text+0x261): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
relations.pb.cc:(.text+0x27a): undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::GroundTruth::GetMetadata() const':
relations.pb.cc:(.text+0x301): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::InitDefaults()':
relations.pb.cc:(.text+0x3a1): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::(anonymous namespace)::AddDescriptorsImpl()':
relations.pb.cc:(.text+0x409): undefined reference to `google::protobuf::MessageFactory::InternalRegisterGeneratedFile(char const*, void (*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&))'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::AddDescriptors()':
relations.pb.cc:(.text+0x471): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::(anonymous namespace)::protobuf_AssignDescriptors()':
relations.pb.cc:(.text+0x58d): undefined reference to `google::protobuf::internal::AssignDescriptors(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, google::protobuf::internal::MigrationSchema const*, google::protobuf::Message const* const*, unsigned int const*, google::protobuf::MessageFactory*, google::protobuf::Metadata*, google::protobuf::EnumDescriptor const**, google::protobuf::ServiceDescriptor const**)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::descriptor()':
relations.pb.cc:(.text+0x701): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::GroundTruth::descriptor()':
relations.pb.cc:(.text+0x8f1): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::protobuf_cartographer_2fground_5ftruth_2fproto_2frelations_2eproto::TableStruct::InitDefaultsImpl()':
relations.pb.cc:(.text+0xa69): undefined reference to `google::protobuf::internal::OnShutdownDestroyMessage(void const*)'
relations.pb.cc:(.text+0xa80): undefined reference to `google::protobuf::internal::OnShutdownDestroyMessage(void const*)'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const':
relations.pb.cc:(.text+0x1c41): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::GroundTruth::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const':
relations.pb.cc:(.text+0x1d1a): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::ByteSizeLong() const':
relations.pb.cc:(.text+0x1d58): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::GroundTruth::ByteSizeLong() const':
relations.pb.cc:(.text+0x1e2c): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::SubmapId::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
pose_graph.pb.cc:(.text+0x93): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(pose_graph.pb.cc.o):pose_graph.pb.cc:(.text+0x113): more undefined references to `google::protobuf::internal::proto3_preserve_unknown_' follow
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::PoseGraph_Constraint::GetMetadata() const':
pose_graph.pb.cc:(.text+0x671): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2fpose_5fgraph_2eproto::(anonymous namespace)::protobuf_RegisterTypes(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
pose_graph.pb.cc:(.text+0x711): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
pose_graph.pb.cc:(.text+0x72a): undefined reference to `google::protobuf::internal::RegisterAllTypes(google::protobuf::Metadata const*, int)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::PoseGraph_LandmarkPose::GetMetadata() const':
pose_graph.pb.cc:(.text+0x7b1): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::NodeId::GetMetadata() const':
pose_graph.pb.cc:(.text+0x851): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::SubmapId::GetMetadata() const':
pose_graph.pb.cc:(.text+0x8f1): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::PoseGraph::GetMetadata() const':
pose_graph.pb.cc:(.text+0x991): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2fpose_5fgraph_2eproto::InitDefaults()':
[... cut ...]
libcartographer.a(trajectory.pb.cc.o): In function `cartographer::mapping::proto::Trajectory::InternalSerializeWithCachedSizesToArray(bool, unsigned char*) const':
trajectory.pb.cc:(.text+0x27b2): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'
libcartographer.a(trajectory.pb.cc.o): In function `cartographer::mapping::proto::Trajectory_Node::ByteSizeLong() const':
trajectory.pb.cc:(.text+0x27f8): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'

```

I have already intalled protobuf 3.4.1, anything wrong? does your cmakelists file link that???????

zchao9456, 2018-12-27:
> rtographer.a(relations.pb.cc.o): In function `cartographer::ground_truth::proto::Relation::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const': relations.pb.cc:(.text+0x133): [...]

Hello, have you solved this problem? I have the same problem. Can you help me? Thank you.

MichaelGrupp, 2019-01-02:
This will probably fixed by #1491

ojura, 2019-01-02:
My guess would be that this is due to Protobuf 3 conflicting with system Protobuf 2 from Ubuntu 16.04 repositories. The proper way to prevent this from happening is to change Protobuf’s `CMAKE_INSTALL_PREFIX`  (and `CMAKE_PREFIX_PATH` when building Cartographer) to e.g. something in /home or in /opt to prevent this from happening. Or simply use Ubuntu 18.04 which comes with Protobuf 3.

ToxicYan, 2019-03-06:
i meet the same question

ojura, 2019-03-06:
@ToxicYan can you try commenting out the line `find_package(Protobuf 3.0.0 CONFIG QUIET)` in `cartographer/CMakeLists.txt`

ToxicYan, 2019-03-07:
@jinfagang @gongyue666 @zchao9456 

I got the solution .
git the protobuf v3.4.1

cd protobuf
mkdir build
cd build
cmake -G Ninja \
  -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
  -DCMAKE_BUILD_TYPE=Release \
  -Dprotobuf_BUILD_TESTS=OFF \
  ../cmake
ninja
sudo ninja install

then

catkin_make_isolated --install --use-ninja

jinfagang, 2019-03-07:
@ToxicYan Why protobuf 3.6.1 does not work?

ToxicYan, 2019-03-07:
https://google-cartographer.readthedocs.io/en/latest/index.html#getting-started-without-ros

you can try 3.6.1
but required on the official website is the vision 3.4.1

jinfagang, 2019-03-07:
@ToxicYan  This error is protobuf 3.6.1 comes

ToxicYan, 2019-03-07:
Are you sure you make protobuf use the right way ?

gongyue666, 2019-03-08:
thx @ToxicYan I will try this weekends.

mrgransky, 2019-12-25:
For anyone who is struggling, I managed to fix such an error as follows:

Do not run ` $ src/cartographer/scripts/install_proto3.sh` as mentioned in this [link](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html). Instead in file `install_proto3.sh` try to remove:

```
mkdir build
cd build
cmake -G Ninja \
  -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
  -DCMAKE_BUILD_TYPE=Release \
  -Dprotobuf_BUILD_TESTS=OFF \
  ../cmake
ninja
sudo ninja install
```
then run ` $ src/cartographer/scripts/install_proto3.sh` in terminal. After that, do the following in terminal one by one:

```
$ ./autogen.sh
$ ./configure --prefix=/usr/local/ CC=/usr/bin/gcc
$ make
$ make check
$ sudo make install
$ sudo ldconfig  # refresh shared library cache
$ protoc --version # just to ensure you have libprotoc 3.4.0 installed.
```
Detail of my system: `$ lsb_release`
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
```

closed 2020-07-09 by wohe