cartographer-project/cartographer #1484: eigen cannot have the rosdep key resolved

p1n0cch10, 2018-12-19:
Hi all,

I am installing cartographer on ROS Indigo and Ubuntu 14.04, following the instruction [here](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html).

However it said eigen cannot have the rosdep key resolved when I executed `rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y`. The log is listed below.

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer: No definition of [eigen] for OS version []
cartographer_ros: No definition of [tf2_eigen] for OS [debian]
ceres-solver: No definition of [eigen] for OS version []
```

Actually, I have the eigen lib installed in `/usr/include/eigen2` and `/usr/include/eigen3`.

I found an issue [here](https://github.com/googlecartographer/cartographer/issues/1121). I followed his method, but the problem remained.

Can anyone tell me how to resolve the problem? Many thanks.

p1n0cch10, 2018-12-20:
well, I find a solution, although I do not know why.

`rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y --os=ubuntu:trusty`

SohaibS2E, 2019-10-21:
``` sudo apt-get install -y google-mock libboost-all-dev  libeigen3-dev libgflags-dev libgoogle-glog-dev liblua5.2-dev libprotobuf-dev  libsuitesparse-dev libwebp-dev ninja-build protobuf-compiler python-sphinx  ros-indigo-tf2-eigen ```