cartographer-project/cartographer #1487: Pose Estimate Doesn't Change when Driving Forward

jimmyw404, 2018-12-28:
I'm attempting to utilize Cartographer to process rosbags containing data recorded from my simulation environment, ANVEL.

A file with the rosbag and all the configuration files I use is here:
https://drive.google.com/file/d/1JBAeskys9FSG5IG1XFNLAcl-O84Em1x7/view?usp=sharing

The particular rosbag I'm having trouble with records an IMU and point cloud when driving straight toward a wall in a simple environment, shown below:
Start:
[image]
Finish:
[image]

When I run this motion live, rviz shows an accurate point cloud representation:
[image]

When I process the recorded data with Cartographer, I see the same point cloud but they appear to move toward the platform which is locked at position 0,0. It's as if the robot is stationary and the environment is moving toward it. You can see how the environment looks smeared.
[image]

The rosbag only contains Imu and PointCloud2 data:
```
j@j:/media/Data/rosbags$ rosbag info ANVEL_StraightWall_Fast.bag 
path:        ANVEL_StraightWall_Fast.bag
version:     2.0
duration:    19.8s
start:       Dec 28 2018 14:25:24.19 (1546025124.19)
end:         Dec 28 2018 14:25:43.97 (1546025143.97)
size:        7.2 MB
messages:    2346
compression: none [10/10 chunks]
types:       sensor_msgs/Imu         [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      Anvel_IMU_Data     1955 msgs    : sensor_msgs/Imu        
             Anvel_Lidar_Data    391 msgs    : sensor_msgs/PointCloud2

```

The output of cartographer_rosbag_validate seems nominal:
```
j@j:~/cartws/src$ cartographer_rosbag_validate -bag_filename /media/Data/rosbags/ANVEL_StraightWall_Fast.bag 
E1228 16:05:08.251893 15727 rosbag_validate_main.cc:190] Sensor with frame_id "lidar" sends exactly the same range measurements multiple times. Range data at time 1546025124.272706000 equals preceding data with 1081 points.
E1228 16:05:08.252557 15727 rosbag_validate_main.cc:190] Sensor with frame_id "lidar" sends exactly the same range measurements multiple times. Range data at time 1546025124.323287000 equals preceding data with 1081 points.
E1228 16:05:08.252638 15727 rosbag_validate_main.cc:190] Sensor with frame_id "lidar" sends exactly the same range measurements multiple times. Range data at time 1546025124.373850000 equals preceding data with 1081 points.
I1228 16:05:08.269693 15727 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "Anvel_IMU_Data" (frame_id: "imu"):
Count: 1954  Min: 0.008976  Max: 0.011418  Mean: 0.010122
[0.008976, 0.009221)	                    	Count: 20 (1.023541%)	Total: 20 (1.023541%)
[0.009221, 0.009465)	                   #	Count: 60 (3.070624%)	Total: 80 (4.094166%)
[0.009465, 0.009709)	                  ##	Count: 235 (12.026612%)	Total: 315 (16.120777%)
[0.009709, 0.009953)	                  ##	Count: 236 (12.077789%)	Total: 551 (28.198566%)
[0.009953, 0.010197)	            ########	Count: 742 (37.973389%)	Total: 1293 (66.171951%)
[0.010197, 0.010441)	                 ###	Count: 272 (13.920164%)	Total: 1565 (80.092117%)
[0.010441, 0.010685)	                 ###	Count: 288 (14.738997%)	Total: 1853 (94.831116%)
[0.010685, 0.010930)	                   #	Count: 62 (3.172978%)	Total: 1915 (98.004097%)
[0.010930, 0.011174)	                    	Count: 33 (1.688843%)	Total: 1948 (99.692940%)
[0.011174, 0.011418]	                    	Count: 6 (0.307062%)	Total: 1954 (100.000000%)
I1228 16:05:08.269810 15727 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "Anvel_Lidar_Data" (frame_id: "lidar"):
Count: 390  Min: 0.049688  Max: 0.051267  Mean: 0.050608
[0.049688, 0.049846)	                    	Count: 1 (0.256410%)	Total: 1 (0.256410%)
[0.049846, 0.050004)	                    	Count: 2 (0.512821%)	Total: 3 (0.769231%)
[0.050004, 0.050162)	                   #	Count: 16 (4.102564%)	Total: 19 (4.871795%)
[0.050162, 0.050320)	                   #	Count: 23 (5.897436%)	Total: 42 (10.769231%)
[0.050320, 0.050478)	                  ##	Count: 45 (11.538462%)	Total: 87 (22.307692%)
[0.050478, 0.050635)	             #######	Count: 141 (36.153847%)	Total: 228 (58.461540%)
[0.050635, 0.050793)	               #####	Count: 96 (24.615385%)	Total: 324 (83.076920%)
[0.050793, 0.050951)	                   #	Count: 26 (6.666667%)	Total: 350 (89.743591%)
[0.050951, 0.051109)	                   #	Count: 28 (7.179487%)	Total: 378 (96.923080%)
[0.051109, 0.051267]	                   #	Count: 12 (3.076923%)	Total: 390 (100.000000%)
```

The transform information is described in an urdf file:
P3-AT_SnapshotLidar.urdf

The frame_id of the lidar is "base_link" and the child_frame_id is "lidar". 
The frame_id of the point cloud produced by the lidar is "lidar". 
The frame_id of the imu is "base_link" and the child_frame_id is "imu".
The frame_id of the imu data is "imu".

The output of tf view_frames during operation seems correct:
[image]

and matches view_frames during live operation of ANVEL:
[image]

The imu produces data at 100hz. The lidar is a planar lidar that produces a 1081 points at 20hz and with 270 degrees field of view. The point cloud itself is an instant snapshot of the environment (rather than rotating over time) to provide the best possible point cloud for Cartographer to use.

When processing the data I looked at the middle point in the point cloud (straight ahead of the robot) to verify it looks correct. It starts at [x:14.50, y:-0.03, z:0.00] (so 14.5meters from the lidar) and finishes at [x:2.90, y:-0.01,  z:0.00] (so 2.9 meters from the lidar) which looks correct.

Looking at the X accel in rqt_plot everything looks correct. I see the initial acceleration and the terminating deceleration:
[image]

I also looked at the delta between the initial_ceres_pose and the pose_observation output of the ceres_scan_matcher in (In LocalTrajectoryBuilder2D:ScanMatch) and charted the X axis of that delta (that I call the adjX) and an integration of that signal (that I call CumAdjX) over time, along with X Acceleration:
[image]

As far as I can tell the scan matching does a very good job of telling the robot it's not moving, because the cumulative adjustment averages to 0 over time.

There must be something very simple that I'm missing with my transforms / coordinate frames. The museum demo bag files all work beautifully. Any advice on what I could be doing wrong or what I can try is welcome.

gaschler, 2019-01-02:
There are several transforms published by more than one node.
Drop robot_state_publisher if anvel already publishes base_link->lidar.
Also, it is problematic that anvel publishes odom->base_link because this is also Cartographer's output, so you have frames fighting.

jimmyw404, 2019-01-02:
> There are several transforms published by more than one node.
> Drop robot_state_publisher if anvel already publishes base_link->lidar.
> Also, it is problematic that anvel publishes odom->base_link because this is also Cartographer's output, so you have frames fighting.

Thank you for your reply. I am still new to ROS, but if I record only the imu and PointCloud2 in the rosbag and replay them in my launch command, won't the transforms published by ANVEL (Which isn't running when I play the rosbag files) be irrelevant? Specifically, when I playback the rosbag:
```
rosbag play ANVEL_StraightWall_Fast.bag 
```

and check what is published it via:
```
rostopic list
```

I get the output:
```
j@j:~/.local/share/Trash/files/build$ rostopic list
/Anvel_IMU_Data
/Anvel_Lidar_Data
/clock
/rosout
/rosout_agg
```
And when I request the TF frames via:
```
rosrun tf view_frames
```

I get this pdf:
[img]https://i.imgur.com/yGVy29P.png[/img]

Thank you again for your help.

jimmyw404, 2019-01-02:
Update: I spent Monday and today adding the ability to generate MultiEchoLaserScan messages from ANVEL and recreating, as much as possible, the kind of robot configuration that is described in the demo_backpack_2d.launch file so that I could use that set of configuration files with my own bag.

After doing so, I got the exact same behavior shown in the above images (walls move, robot jitters around 0,0). I noticed the packet rate of the demo bag files was higher than mine (lidar at 40hz vs 20hz), so I increased my sample rate and discovered that the pose updated correctly only when it approached the end of the path! So I tried to go diagonally in my environment instead of orthogonally and the performance of Cartographer was quite good. I went back to my original setup (that produced PointCloud2 and IMU data), increased the data rate and travelled diagonally and discovered that worked well also.

Below is the Diagonal run and a short "tour" of my environment:
[image]
[image]

I still get poor performance when travelling straight down one of the rooms (Too many walls that are parallel with the direction of travel?) and when I slowly turn in place in a circle, but I think that belongs in a new issue and not this one.

I'm going to close this with the solution being:
1. Don't travel in parallel to the direction of travel?
2. Increase data rate of sensors?

closed 2019-01-02 by jimmyw404

ojura, 2019-01-02:
FYI Cartographer can handle lower sample rate lidars just fine (we’ve done vehicle positioning experiments with a Sick NAV which spins around 8 Hz, with lateral repeatability under a centimeter). Your problem is elsewhere. Try changing the translation weight of the local trajectory builder (see the tuning guide for more info).