cartographer-project/cartographer #1505: Cannot test Cloud locally

ojura, 2019-01-17:
If I try 
```
docker build . -t cartographer:trusty -f Dockerfile.trusty --build-arg cc=gcc --build-arg cxx=g++
```
I get one failing test, like this:

```
Running main() from gmock_main.cc
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from LocalTrajectoryUploaderTest
[ RUN      ] LocalTrajectoryUploaderTest.HandlesInvalidUplink
E0117 19:38:40.493330864    4009 google_default_credentials.cc:283] Could not create google default credentials.
E0117 19:38:40.493455476    4009 google_default_credentials.cc:288] grpc_google_default_credentials_create: {"created":"@1547753920.394662388","description":"Failed to create Google credentials","file":"src/core/lib/security/credentials/google_default/google_default_credentials.cc","file_line":225,"referenced_errors":[{"created":"@1547753920.394665482","description":"creds_path unset","file":"src/core/lib/security/credentials/google_default/google_default_credentials.cc","file_line":168},{"created":"@1547753920.394701463","description":"Failed to load file","file":"src/core/lib/iomgr/load_file.cc","file_line":69,"filename":"/root/.config/gcloud/application_default_credentials.json","referenced_errors":[{"created":"@1547753920.394697166","description":"OS Error","errno":2,"file":"src/core/lib/iomgr/load_file.cc","file_line":43,"os_error":"No such file or directory","syscall":"fopen"}]}]}
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0117 19:38:40.493795  4009 local_trajectory_uploader.cc:123] Connecting to uplink invalid-uplink-address:50051
E0117 19:38:40.493837579    4009 channel_connectivity.cc:47] grpc_channel_check_connectivity_state called on something that is not a client channel, but 'lame-client'
```

I wonder how come this passes on Travis, and not on my box (Ubuntu 18.04, Docker version 18.09.1, build 4c52b90)?

xuguangyun, 2019-09-04:
same to me..

xuguangyun, 2019-09-04:
i donot know why but change to  grpc::InsecureChannelCredentials()  slove this error

rzaddem, 2019-11-11:
@xuguangyun Would you please be more specific? I'm running into the same issue. Where can we make this change?