cartographer-project/cartographer #1509: Could cartographer deal with hijack situation ?

PeterTdd, 2019-01-25:
After the robot travels a certain amount of time,  someone pick it up and put it to somewhere the robot travelled before (the place could be several meters away). Could cartographer relocalize the robot's position at this hijack situation ?

ojura, 2019-01-25:
Yes, provided that you finish building the first trajectory before the hijacking happens and start a new one after hijacking is finished. After driving around for a while, it will figure out where it is with respect to the first trajectory.

PeterTdd, 2019-01-25:
@ojura , thank you for your answer.
I find that I can call FinishTrajectory() in map_builder to finish the first trajectory.
How about start a new trajectory ?
Do I need to call AddTrajectoryBuilder ?
What if the hijack happens at the middle of the first trajectory? 
Will it be too long for waiting the FinishTrajectory() to finish the first trajectory and then start a new one?

ojura, 2019-01-25:
Hopefully FinishTrajectory() should not take too long. Have a look at how `cartographer_ros` starts a new trajectory: 
https://github.com/googlecartographer/cartographer_ros/blob/6ee68c22f922c6234c5488f649f1b413e860eae2/cartographer_ros/cartographer_ros/map_builder_bridge.cc#L124

PeterTdd, 2019-01-28:
Thanks. I will try that

closed 2020-05-29 by wohe