cartographer-project/cartographer #1514: Tsdf tests flaky

gaschler, 2019-02-01:
The tests following tests related to tsdf are flaky:

mapping/2d/tsdf_range_data_inserter_2d_test.cc
RangeDataInserterTest2DTSDF.InsertSmallAnglePointWitNormalProjection
ray_to_pixel_mask.cc:44 Check failed: scaled_begin.y() >= 0 (-165 vs. 0) 

mapping/internal/2d/scan_matching/tsdf_match_cost_function_2d_test.cc:107
Value of: jacobians[0]
Expected: has 3 elements where
element #0 is approximately 0 (absolute error <= 0.001),
element #1 is approximately -1 (absolute error <= 0.001),
element #2 is approximately 0 (absolute error <= 0.001)
  Actual: { 0, 0, 0 }, whose element #1 doesn't match, which is 1 from -1

mapping/internal/2d/scan_matching/tsdf_match_cost_function_2d_test.cc:125
Value of: residuals
Expected: has 1 element that is approximately -0.10000000000000001 (absolute error <= 0.001)
  Actual: { 0 }, whose element #0 doesn't match, which is 0.1 from -0.1

ojura, 2019-02-01:
Looks like #1466 is not specific to windows.

kdaun, 2019-02-04:
I executed the tests ~20 times with varying system loads and was not able to reproduce the failing tests. Can you give some more background under which conditions and with which frequency the tests tend to fail?

gaschler, 2019-02-05:
tsdf_match_cost_function_2d_test flake very rarely under default compile options.

bazel test --runs_per_test=1000 //cartographer:mapping_internal_2d_scan_matching_tsdf_match_cost_function_2d_test --test_output=errors
(fails roughly 1/1000 in my case)

I find this unexpected because the test does not do any input/output so I don't see where the flakiness comes from. (uninitialized memory?)

The tsdf_range_data_inserter_2d_test flake I was not able to reproduce.

ojura, 2019-02-05:
IIRC on Windows every third or so test run fails.

kdaun, 2019-02-10:
@gaschler Uninitialized memory seems like a possible explanation to me as well.

Executing the TSDF related tests with Valgrind yields _"Conditional jump or move depends on uninitialised value(s)"_ and _"Uninitialized value was created by a stack allocation"_ warnings for tsdf_match_cost_function_2d_test.cc and tsdf_range_data_inserter_2d_test.cc. Which are exactly the flaky tests.
The initialization for `sensor::RangeData range_data;` is missing. I fixed the initialization in https://github.com/googlecartographer/cartographer/pull/1520. 

As I cannot reproduce the flakiness on my system, I cannot evaluate whether the PR fixes the issue.

gaschler, 2019-02-11:
@kdaun Great, thanks for looking into this and finding the root cause!

gaschler, 2019-02-13:
Fixed by https://github.com/googlecartographer/cartographer/pull/1520, closing.
@kdaun Thanks again for the fix!

closed 2019-02-13 by gaschler