cartographer-project/cartographer #1515: Actual 3D pointcloud creation

jattiogbe, 2019-02-04:
Hi,

I am new to Cartographer, I've tested the demos and I was disappointed to see what they call 3D SLAM since I was expecting to see the pointcloud of an actual 3D museum, not just the floor...

I've read lots of issues here and I don't understand all of them. I think the closer one to my goal is [this one (1168)](https://github.com/googlecartographer/cartographer_ros/issues/1168).
I don't understand how to process things explained [here](https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html#first-person-visualization-of-point-clouds). 
My goal is to get a .ply file out of my lidar data (ie bagfiles), I am not "fluent" in this developping  environment words : asset, pipeline.
Can you explain me what should I do ? what should I precisely type, execute ?

thanks.

jattiogbe, 2019-02-04:
I am moving this issue to Cartographer_ROS, so I'm closing.

Semih67, 2019-02-05:
Hey I can explain you what I have done. But I have to tell you that I have errors so it's not working but I think I came to the solution closer:

1.first of all you need a bag file
2.Start your online node for your cartographer
3.type this command: rosservice call /finish_trajectory 0
4.then this: rosservice call /write_state "{filename: 'dir to your bag.pbstream'}" This is the directory where you want to save your pbstream
5. launch your bag file with these -> bag_filenames:= directory and pose_graph_filename:= directory to your pbstream

-So you need to use the assets_writer_backpack_3d.lua which you can find in the config_files of cartographer_ros (or you can write your own script)
-you need a launch file for your assets_writer: they have also one in the launch folder -> assets_writer_backpack_3d.launch 

Nevertheless I couldn't configure it well enough or I'm doin somethin wrong with the bagfile, because I get an error which says: assets_writer_main.cc:53] Check failed: !FLAGS_bag_filenames.empty() -bag_filenames is missing 
REQUIRED process [cartographer_assets_writer-1] has died! 

I'm recording my bagfile with this command: rosbag record -a maybe this is the problem but I'm not sure

MichaelGrupp, 2019-02-05:
To write an `.ply` file, add:
```lua
{
    action = "write_ply",
    filename = "points.ply",
},
```

To write an `.xyz` file, add:
```
{
    action = "write_xyz",
    filename = "points.xyz",
},
```

MichaelGrupp, 2019-02-05:
I agree that it's not very clear how the configuration needs to look like from the code or the documentation, this is something that can be improved.

jattiogbe, 2019-02-05:
Based on  [assets_writer_backpack_2d.lua.](https://github.com/googlecartographer/cartographer_ros/blob/44459e18102305745c56f92549b87d8e91f434fe/cartographer_ros/configuration_files/assets_writer_backpack_2d.lua), I got this so I can get a .ply file: 
```
    {
      action = "write_ply",
      filename = "points.ply",
},
```
I edited **assets_writer_backpack_3d.lua** with this code and **assets_writer_backpack_3d.launch** to call my new .lua file.
I've got a 1.7 Go .ply file that I open with CloudCompare (maybe I'll try point_cloud_viewer
later) , it seems fine:

[image]

I will try the xyz format and let you know, I need to be able to reduce the size of the file also!

MichaelGrupp, 2019-02-05:
> I need to be able to reduce the size of the file also!

You should run a filter on the raw pointcloud, in CloudCompare you can use for example an SOR filter (https://www.cloudcompare.org/doc/wiki/index.php?title=SOR_filter).

There is a "fixed ratio sampler" that you can use to filter directly in the assets writer, but it's not very smart. It just drops a fixed ratio of points:
```lua
    {
      action = "fixed_ratio_sampler",
      sampling_ratio = 0.5,
    },
```
But I would rather recommend to use something like the SOR filter instead of this.

Semih67, 2019-02-11:
Hey @jattiogbe how did you get the pointcloud? What were your first steps?
What I am doing is:
1. record the bagfile
2. stop recording
3. launch the bagfile
4. then finishing trajectory
5. writing state
6 stop bag file
7. launch assets_writer with the bag file and pbstream

But this is not working for me

jattiogbe, 2019-02-11:
Hi @Semih67 ,
I can't get a proper pointcloud of my own, I don't even passe the rosbag_validate step, I don't know what's wrong with my recordings.

- I am in Lidar_mode 1024x20
-  I do : 
`roslaunch ouster_ros os1.launch os1_hostname:=os1-991838000761.local os1_udp_dest:=192.168.1.7 viz:=true scan_dur_ns:=50000000`
and 
`
rosbag record /os1_node/imu_packets /os1_node/lidar_packets`

- then

 when I want to validate :
`
rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ~/cat_ws/demo_2019-02.bag `
which returns nothing.

I guess I'm stuck there until I manage to validate my bagsfiles.

Semih67, 2019-02-12:
It woked for me now. I was publishing a tf static from world to map. And this was the issue. When I deleted this it worked

jattiogbe, 2019-02-12:
Are you using default launch files or the modified ones we can find on github? 
What lidar model do you have ? I have an Ouster OS1. 
I resolved my problem of validating bagfiles, I'm nowhere further this step.

What file did you delete the "tf static from world to map" line ?

Semih67, 2019-02-18:
I'm using a tof camera not a lidar. I'm recording it like this: rosbag record /tf /tf_static /imu /scan_matched_points2

And yes this -> "tf static from world to map" was not necessary and I deleted it.

Duke-Allen, 2019-02-28:
@jattiogbe 
Hello, how do you get the pointcloud? I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. 
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem about it?

closed 2020-08-26 by wohe