cartographer-project/cartographer #1518: TSDF fails in deutsch museum rosbag

jihoonl, 2019-02-07:
[image]

I recently played Deutsch museum rosbag with TSDF submap enabled to see the performance of TSDF compare to probability grid. And I observed this drifting of map. It seems like some pose transformation are applied with wrong time stamp. Am I using inappropriate configuration?

```
# Added in backpack_2d.lua
TRAJECTORY_BUILDER_2D.submaps.grid_options_2d.grid_type = "TSDF"
TRAJECTORY_BUILDER_2D.submaps.grid_options_2d.resolution = 0.05
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.range_data_inserter_type = "TSDF_INSERTER_2D"
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.tsdf_range_data_inserter.truncation_distance = 0.05
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.tsdf_range_data_inserter.maximum_weight = 10.
```

kdaun, 2019-02-07:
The quality of TSDF results are sensitive to settings.
See https://github.com/googlecartographer/cartographer_ros/issues/1058 for an example of a working configuration of the Deutsches Museum dataset.

jihoonl, 2019-02-11:
Thanks, @kdaun. I got much decent result compare to the video above. 
Is there any reason that it isn't the default parameter of tsdf range data inserter?

kdaun, 2019-02-13:
> Is there any reason that it isn't the default parameter of tsdf range data inserter?
This seems reasonable for the parameters not affecting the ProbabilityGrid. Furthermore, it would be useful to add the configuration as a TSDF demo setup to cartographer_ros.