cartographer-project/cartographer #1519: Fix Bazel incompatible changes

laurentlb, 2019-02-07:
According to our Bazel CI (https://buildkite.com/bazel/bazel-at-release-plus-incompatible-flags/builds/95), the project doesn't build with the following Bazel flags:

* `--incompatible_new_actions_api`
* `--incompatible_disable_deprecated_attr_params`
* `--incompatible_disallow_load_labels_to_cross_package_boundaries`

laurentlb, 2019-03-06:
Any update on this?

Based on our latest CI run (https://buildkite.com/bazel/bazelisk-plus-incompatible-flags/builds/28#5b3ff861-39dc-4ee4-a79c-3ea1b175b6ea), migration is needed for the following flags:
  * --incompatible_bzl_disallow_load_after_statement
  * --incompatible_disable_deprecated_attr_params
  * --incompatible_disallow_load_labels_to_cross_package_boundaries
  * --incompatible_new_actions_api

Let me know if you need help with it.

meteorcloudy, 2019-03-21:
@gaschler Please help resolving this issue, we have to disable Cartographer in Bazel downstream for now. Sorry for that~
https://github.com/bazelbuild/continuous-integration/pull/582

cschuet, 2019-03-30:
@meteorcloudy Should be fixed now. Can you re-enable us in Bazel downstream please?

meteorcloudy, 2019-04-01:
After #1553 , there is one test failing on Bazel CI, can you take a look?
https://buildkite.com/bazel/cartographer/builds/1619#4e1c1f09-6ccf-43fd-802c-d096fb7ec4c8

gaschler, 2019-04-01:
> //cartographer/cloud:internal_client_server_test                         FAILED in 5 out of 5 in 21.1s
> [ RUN      ] ClientServerTestByGridType/ClientServerTestByGridType.LocalSlam2DUplinkServerRestarting/0
> [KOptimizing: Done.     
> I0401 08:09:09.712164  3127 map_builder_server.cc:158] Starting SLAM thread.
> I0401 08:09:09.712529  2658 map_builder_stub.cc:52] Connecting to SLAM process at 0.0.0.0:50051 with client_id CLIENT_ID
> I0401 08:09:09.725284  3111 local_trajectory_uploader.cc:148] Trying to re-connect to uplink...
> I0401 08:09:09.725651  3111 local_trajectory_uploader.cc:157] Uplink channel ready, trying recovery.
> I0401 08:09:09.725672  3111 local_trajectory_uploader.cc:160] LocalTrajectoryUploader tries to recover with next submap.
> F0401 08:09:09.734764  2658 pose_graph_stub.cc:93] Check failed: client.Write(request) 
> *** Check failure stack trace: ***
> external/bazel_tools/tools/test/test-setup.sh: line 310:  2658 Aborted    

I think the solution is to disable the "LocalSlam2DUplinkServerRestarting", "LocalSlam3DUplinkServerRestarting" tests (but not the entire client_server_test.cc)
They only fail on a rarely used feature.

meteorcloudy, 2019-04-01:
@gaschler Can you help disable the test? Thanks!

cschuet, 2019-04-02:
Test should be disabled now.

PR #1553 (2019-03-30): Update bazel dependencies
Update protobuf grpc and bazel-rules and added bazel-skylib (required by
protobuf 3.7.1).

Closes #1519
commit b1edbea8 (2019-03-30): Update bazel dependencies (#1553)

* Update dependencies

Update protobuf grpc and bazel-rules and added bazel-skylib (required by
protobuf 3.7.1).

Closes #1519
closed 2021-01-27 by MichaelGrupp