cartographer-project/cartographer #1523: High heap memory consumption of OverlappingSubmapsTrimmer

MichaelGrupp, 2019-02-22:
The overlapping submaps trimmer in its current implementation requires a lot of dynamic memory, mainly to do the coverage computation. It computes a `SubmapCoverageGrid2D` in which each grid cell contains a vector of the submap IDs that cover that coordinate at point in time - `using StoredType = std::vector<std::pair<SubmapId, common::Time>>`. This can lead to problems especially when dealing with large maps, where the coverage grid grows very large.

To illustrate, I ran a heap profiler ([massif](http://valgrind.org/docs/manual/ms-manual.html)) on an offline node that loads a pose graph from a .pbstream, runs final optimization with overlapping submaps trimmer and exits. The map was rather small/medium size to keep profiling time down but you get the point.

The width of the colored bands in the plot show how the data structures used in the trimmer grow while the submap coverage is computed.

[image]

MichaelGrupp, 2019-02-22:
We can collect some ideas for improving the memory efficiency in this issue, e.g.:
* use a low-resolution coverage grid
* use a different approach for computing overlap that doesn't create a separate grid

pifon2a, 2019-02-25:
@MichaelGrupp thank you for finding this. With what parameters do you use the trimmer? If you keep a small number of submaps per pixel, we can clean up `vector<pair<SubmapId, Time>>` in the loop.

MichaelGrupp, 2019-02-25:
I used:
```lua
POSE_GRAPH.overlapping_submaps_trimmer_2d = {
   fresh_submaps_count = 3,
   min_covered_area = 1.0,
   min_added_submaps_count = 5
}
```

and ran
```
valgrind --tool=massif cartographer_offline_node --load_state_filename ... --configuration_directory ... --configuration_basenames ... --load_frozen_state=false
```