cartographer-project/cartographer #1526: cartographer/mapping/proto/2d/cell_limits.pb.h: No such file or directory

tyaga08, 2019-03-04:
I am trying to use cartographer to generate a 2D probability grid. When I am trying to add the header file to my code, it is throwing this error. I am new to cartographer and I am unable to find any good resource for learning how to use it. Please help me

In file included from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/map_limits.h:26:0,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/grid_2d.h:22,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/probability_grid.h:23,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node1.cpp:1:
/home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/xy_index.h:28:58: fatal error: cartographer/mapping/proto/2d/cell_limits.pb.h: No such file or directory
compilation terminated.
In file included from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/map_limits.h:26:0,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/grid_2d.h:22,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/probability_grid.h:23,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node2.cpp:1:
/home/tyagaraja/catkin_ws/src/Mini_SLAM/../../../cartographer_ws/src/cartographer/cartographer/mapping/2d/xy_index.h:28:58: fatal error: cartographer/mapping/proto/2d/cell_limits.pb.h: No such file or directory
compilation terminated.
Mini_SLAM/CMakeFiles/node1.dir/build.make:62: recipe for target 'Mini_SLAM/CMakeFiles/node1.dir/src/node1.cpp.o' failed
make[2]: *** [Mini_SLAM/CMakeFiles/node1.dir/src/node1.cpp.o] Error 1
CMakeFiles/Makefile2:2047: recipe for target 'Mini_SLAM/CMakeFiles/node1.dir/all' failed
make[1]: *** [Mini_SLAM/CMakeFiles/node1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Mini_SLAM/CMakeFiles/node2.dir/build.make:62: recipe for target 'Mini_SLAM/CMakeFiles/node2.dir/src/node2.cpp.o' failed
make[2]: *** [Mini_SLAM/CMakeFiles/node2.dir/src/node2.cpp.o] Error 1
CMakeFiles/Makefile2:2010: recipe for target 'Mini_SLAM/CMakeFiles/node2.dir/all' failed
make[1]: *** [Mini_SLAM/CMakeFiles/node2.dir/all] Error 2
Makefile:138: recipe for target 'all' failed

pifon2a, 2019-03-05:
@tyaga08 If `*pb.h` header is not found, then it means that `cartographer/mapping/proto/2d/cell_limits.proto` was not compiled. Could you please send a link to your branch?

tyaga08, 2019-03-05:
@pifon2a  There were couple of things that I was doing wrong. I had installed ros-kinetic-cartographer as well as the cartographer_ros from the documentation. Secondly, I also had not written my CMakeLists.txt properly it seems. So I uninstalled the ros-kinetic package and also changed my CMakeLists.txt and it worked. Thank you for your help.

closed 2019-03-05 by tyaga08