cartographer-project/cartographer #1527: older version of protoc incompatible with the protocol buffer headers

tyaga08, 2019-03-04:
I have installed cartographer using apt-get install ros-kinetic-cartographer
when I am adding this to my package and building, I am getting this error. What should I do? Please help.

In file included from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:23:0,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node2.cpp:1:
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:27:0,
                 from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:23,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node2.cpp:1:
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /opt/ros/kinetic/include/cartographer/common/lua_parameter_dictionary.h:25:0,
                 from /opt/ros/kinetic/include/cartographer/transform/rigid_transform.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:24,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node2.cpp:1:
/opt/ros/kinetic/include/cartographer/common/lua.h:20:19: fatal error: lua.hpp: No such file or directory
compilation terminated.
In file included from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:23:0,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node1.cpp:1:
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
In file included from /opt/ros/kinetic/include/cartographer/sensor/proto/sensor.pb.h:27:0,
                 from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:23,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node1.cpp:1:
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is
  ^
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
 #error incompatible with your Protocol Buffer headers.  Please
  ^
/opt/ros/kinetic/include/cartographer/transform/proto/transform.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
 #error regenerate this file with a newer version of protoc.
  ^
Mini_SLAM/CMakeFiles/node2.dir/build.make:62: recipe for target 'Mini_SLAM/CMakeFiles/node2.dir/src/node2.cpp.o' failed
make[2]: *** [Mini_SLAM/CMakeFiles/node2.dir/src/node2.cpp.o] Error 1
In file included from /opt/ros/kinetic/include/cartographer/common/lua_parameter_dictionary.h:25:0,
                 from /opt/ros/kinetic/include/cartographer/transform/rigid_transform.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/point_cloud.h:24,
                 from /opt/ros/kinetic/include/cartographer/sensor/compressed_point_cloud.h:25,
                 from /opt/ros/kinetic/include/cartographer/sensor/range_data.h:21,
                 from /opt/ros/kinetic/include/cartographer/mapping/trajectory_node.h:25,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/map_limits.h:26,
                 from /opt/ros/kinetic/include/cartographer/mapping_2d/probability_grid.h:31,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/include/include_files/include_files.h:11,
                 from /home/tyagaraja/catkin_ws/src/Mini_SLAM/src/node1.cpp:1:
/opt/ros/kinetic/include/cartographer/common/lua.h:20:19: fatal error: lua.hpp: No such file or directory
CMakeFiles/Makefile2:2010: recipe for target 'Mini_SLAM/CMakeFiles/node2.dir/all' failed
compilation terminated.
make[1]: *** [Mini_SLAM/CMakeFiles/node2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Mini_SLAM/CMakeFiles/node1.dir/build.make:62: recipe for target 'Mini_SLAM/CMakeFiles/node1.dir/src/node1.cpp.o' failed
make[2]: *** [Mini_SLAM/CMakeFiles/node1.dir/src/node1.cpp.o] Error 1
CMakeFiles/Makefile2:2047: recipe for target 'Mini_SLAM/CMakeFiles/node1.dir/all' failed
make[1]: *** [Mini_SLAM/CMakeFiles/node1.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

pifon2a, 2019-03-05:
Please, take a look at the [ROS compiling instructions](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html) and [installation instructions](https://google-cartographer.readthedocs.io/en/latest/).

tyaga08, 2019-03-05:
@pifon2a Thank you

closed 2019-03-05 by tyaga08