cartographer-project/cartographer #1528: Unexpected process interruption

ToxicYan, 2019-03-05:
When I roslaunch the museum's data package, the map is almost built and it gets stuck.

[ INFO] [1551773099.308228884, 1432648929.449913835]: I0305 16:04:59.000000 21787 constraint_builder_2d.cc:266] Node (0, 37571) with 104 points on submap (0, 418) differs by translation 0.02 rotation 0.000 with score 64.0%.
[ INFO] [1551773099.314684341, 1432648929.449913835]: I0305 16:04:59.000000 21784 constraint_builder_2d.cc:266] Node (0, 37561) with 105 points on submap (0, 418) differs by translation 0.01 rotation 0.000 with score 62.4%.
[ INFO] [1551773099.318799266, 1432648929.449913835]: I0305 16:04:59.000000 21786 constraint_builder_2d.cc:266] Node (0, 37578) with 111 points on submap (0, 418) differs by translation 0.01 rotation 0.001 with score 66.2%.
[ INFO] [1551773099.329978863, 1432648929.449913835]: I0305 16:04:59.000000 21786 constraint_builder_2d.cc:266] Node (0, 37588) with 101 points on submap (0, 418) differs by translation 0.02 rotation 0.001 with score 67.6%.
[ INFO] [1551773099.335385427, 1432648929.449913835]: I0305 16:04:59.000000 21787 constraint_builder_2d.cc:266] Node (0, 37581) with 101 points on submap (0, 418) differs by translation 0.02 rotation 0.000 with score 65.2%.
[ INFO] [1551773099.342970882, 1432648929.449913835]: I0305 16:04:59.000000 21785 constraint_builder_2d.cc:266] Node (0, 37575) with 105 points on submap (0, 418) differs by translation 0.02 rotation 0.000 with score 63.1%.
[ INFO] [1551773099.348147794, 1432648929.449913835]: I0305 16:04:59.000000 21784 constraint_builder_2d.cc:266] Node (0, 37585) with 110 points on submap (0, 418) differs by translation 0.02 rotation 0.000 with score 64.7%.
[ INFO] [1551773099.350238208, 1432648929.449913835]: I0305 16:04:59.000000 21786 constraint_builder_2d.cc:266] Node (0, 37591) with 112 points on submap (0, 418) differs by translation 0.03 rotation 0.003 with score 63.9%.
[ INFO] [1551773099.367693884, 1432648929.449913835]: I0305 16:04:59.000000 21784 constraint_builder_2d.cc:266] Node (0, 37601) with 112 points on submap (0, 418) differs by translation 0.02 rotation 0.001 with score 62.8%.
[ INFO] [1551773099.377763826, 1432648929.449913835]: I0305 16:04:59.000000 21787 constraint_builder_2d.cc:266] Node (0, 37595) with 105 points on submap (0, 418) differs by translation 0.02 rotation 0.002 with score 64.9%.
[ INFO] [1551773099.378172494, 1432648929.449913835]: I0305 16:04:59.000000 21784 constraint_builder_2d.cc:266] Node (0, 37608) with 103 points on submap (0, 418) differs by translation 0.01 rotation 0.003 with score 76.7%.
[ INFO] [1551773099.386017434, 1432648929.449913835]: I0305 16:04:59.000000 21786 constraint_builder_2d.cc:266] Node (0, 37605) with 106 points on submap (0, 418) differs by translation 0.02 rotation 0.002 with score 72.4%.
[ INFO] [1551773099.388049099, 1432648929.449913835]: I0305 16:04:59.000000 21784 constraint_builder_2d.cc:266] Node (0, 37615) with 106 points on submap (0, 418) differs by translation 0.03 rotation 0.003 with score 67.0%.
[ INFO] [1551773099.389125824, 1432648929.449913835]: I0305 16:04:59.000000 21785 constraint_builder_2d.cc:266] Node (0, 37598) with 105 points on submap (0, 418) differs by translation 0.03 rotation 0.003 with score 64.4%.
[ INFO] [1551773099.397123358, 1432648929.449913835]: I0305 16:04:59.000000 21787 constraint_builder_2d.cc:266] Node (0, 37611) with 105 points on submap (0, 418) differs by translation 0.02 rotation 0.002 with score 67.1%.
[ INFO] [1551773099.399706679, 1432648929.449913835]: I0305 16:04:59.000000 21786 constraint_builder_2d.cc:266] Node (0, 37618) with 107 points on submap (0, 418) differs by translation 0.01 rotation 0.001 with score 70.3%.

Then it didn't respond.
And this is the stable vision.
I hope someone has the same problem as me and knows how to solve it...

The latest vision is also have this issue.And it is displayed "pose_graph_2d.cc:382]Remaining work items in queue: 3564 "

I am very puzzled

pifon2a, 2019-03-05:
@ToxicYan: very interesting. On what machine are you launching Cartographer? Cartographer works like this: after a new trajectory node (lidar scan) was added, it adds various work items to the queue, mostly constraint searches. After the searches are done, the whole pose graph is optimized. 

If the machine is very slow or the settings imply lots of computations, then machine cannot keep up with the queue size.

ToxicYan, 2019-03-06:
@pifon2a Thanks for your attention! I'm launching cartographer on my computer with i5-6300 core and NVIDIA GTX960. So i think this configuration is enough to launch the 2d demo...  

Or do you know what parameters can be set to reduce the amount of calculation? I am going to try other demo data bag .

ToxicYan, 2019-03-06:
I re-attempted an official bag and didn't seem to do the final optimization.

[image]

pifon2a, 2019-03-06:
The machine is definitely powerful enough.
Coud it be that you are running the online node and not the offline? The online node generates the pbstream only if you ask it for using `rosservice call` (see the definition of the service [here](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros_msgs/srv/WriteState.srv)).
You can run the offline node using 
```
roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=<path_to_your_bag_file>
```
This node by default writes the pbstream after it has finished. See more [here](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html).

MichaelGrupp, 2019-03-06:
> ... do you know what parameters can be set to reduce the amount of calculation?

You can find some tips here: https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency

In my experience, you can improve performance already a lot by reducing `POSE_GRAPH.constraint_builder.sampling_ratio`, e.g. to 0.05 (default is 0.3). Usually the quality stays the same even if less constraint searches are done.

ToxicYan, 2019-03-07:
I got the solution ,because the protobuf have no totally make ..

Thank you very much !

closed 2019-03-07 by ToxicYan