cartographer-project/cartographer #1530: NormalEstimation Test Flaky

kdaun, 2019-03-06:
https://github.com/googlecartographer/cartographer/pull/1508#issuecomment-470083540 indicates that the normal estimation tests are flaky.
Test log: https://travis-ci.org/googlecartographer/cartographer/jobs/502507460

pifon2a, 2019-03-07:
I will take a look.

kdaun, 2019-03-07:
It's the same issue as in https://github.com/googlecartographer/cartographer/pull/1520.
There are instances of RangeData that need to be initialized explicitly in
https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/2d/normal_estimation_2d_test.cc
We have to replace
`sensor::RangeData range_data;` by   `auto range_data = sensor::RangeData();`.

pifon2a, 2019-03-08:
solved by https://github.com/googlecartographer/cartographer/pull/1532

closed 2019-03-08 by pifon2a