cartographer-project/cartographer #1533: Question: why tsdf? is it better for scan matching compared to probability grid?

kings-landing, 2019-03-07:


pifon2a, 2019-03-08:
It is comparable in CPU time and matching quality. It is more expensive to insert data in TSDF grid than into a probability grid, but it gives you better convergence in optimization, because the cost function is a bit more regular. Also it is easier to build a mesh out of TSDF grid, which is a nice bonus.

fzheng010, 2019-03-08:
Thanks! If I understand correctly, for 2D SLAM, TSDF is comparable to probability grid, but for 3D SLAM, it has an edge, in terms of convergence property and meshing. Is it right? @pifon2a

pifon2a, 2019-03-08:
For 3D SLAM it should have an edge, but I cannot say for sure, since it is not yet implemented in Cartographer.

closed 2019-03-13 by kings-landing