cartographer-project/cartographer #1534: Questions about mapping, remapping, and localization

grieneis, 2019-03-07:
I'm working on an application where we have robots navigating in warehouses that can be up to 50k sq m.  We have been successfully using cartographer for mapping with a different localization algorithm.  We're now looking into switching to cartographer for localization and after reading through some code and documentation, I have some questions.

If this isn't the correct forum for them, please let me know what would be more appropriate, and I will post there.

Our use case:
A customer purchases our system for the warehouse.  We take a robot to the site and create a map.  We save the data from that mapping session and distribute it to the robots that will be doing work at that site.  Later, the customer may request that we add a new section to the warehouse, so we need to update the map to add it.  Or, they may rearrange a couple shelves so we need to remap that section of the warehouse.  With other localization algorithms such as ros's AMCL, the data store is an image file, so these additions and modifications are pretty easy to do.  However, they seem like they may be more difficult with cartographer.

- What happens if we remove an object from an already mapped area and try to remap it? Do the old submaps that correspond to that area get replaced with the new ones?
- What happens if we remove an object from an already mapped area and use localization only mode?  For example, there was a forklift parked in an aisle when the map was made, but the forklift isn't there when the robots need to localize.
- What does it mean for a trajectory to be "frozen"?  What is the difference between loading a frozen trajectory and a non-frozen trajectory?
- When we want to add a new area to the map, do we need to load the old section of the map frozen or unfrozen?
- What does finalizeTrajectory actually do?  Do we need to call it before writing the .pbstream file to disk?
- Is there anything we need to worry about as the map size increases?  Can you anticipate any issues localizing in a 50k sq. m warehouse?

pifon2a, 2019-03-08:
@grieneis let me answer these questions one by one:

_What does it mean for a trajectory to be "frozen"? What is the difference between loading a frozen trajectory and a non-frozen trajectory?_
Trajectory in Cartographer is a collection of robot poses that are optimized given computed constraints (relative transforms between consecutive nodes computed from wheel odometry, poses of robots in a given submaps, etc.). If a trajectory is "frozen", this means that all optimization variables for trajectory nodes and associated submaps will be constant and won't be optimized. If you load a trajectory as a "non-frozen", then they will be optimized. This is done mostly for localization mode: in localization mode you load a frozen pbstream and then optimize only the trajectory of the robot that you want to localize. The trajectory for this robot will be built, but only a constant number of submaps for this trajectory be kept, the old ones will be trimmed. This is needed to keep complexity of the algorithm more or less constant. Look for `PureLocalizationTrimmer` in the [code](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/pose_graph_trimmer.cc).

_What happens if we remove an object from an already mapped area and try to remap it? Do the old submaps that correspond to that area get replaced with the new ones?_
If you want to remap a part of the map, then you can load a pbstream in non-frozen mode, use "overlapping_submaps_trimmer" then the old submaps for the area that you are remapping should be replaced with the newer ones.

Here are the options for trajectory_builder.lua. You might want to play with them.
```
 overlapping_submaps_trimmer_2d = {
 fresh_submaps_count = 2,
 min_covered_area = 2,
 min_added_submaps_count = 5,
 },
```

_What happens if we remove an object from an already mapped area and use localization only mode? For example, there was a forklift parked in an aisle when the map was made, but the forklift isn't there when the robots need to localize._
Localization mode should be quite robust with respect to such small changes of the map. 

_When we want to add a new area to the map, do we need to load the old section of the map frozen or unfrozen?_
Unfrozen, because you want to jointly optimize the whole map.

_What does finalizeTrajectory actually do?_
 Stops waiting for sensor data for the given trajectory. Sensor data in Cartographer is collated before processing.

_Do we need to call it before writing the .pbstream file to disk?_
It is not necessary. You need to call [WriteState](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros_msgs/srv/WriteState.srv) if you want to write a pbstream file while running the online node.

_Is there anything we need to worry about as the map size increases? Can you anticipate any issues localizing in a 50k sq. m warehouse?_
Definitely. You might need to reduce the number of global searches, etc to reduce the load.

D4YDR34MPH4N70M, 2019-04-10:
The legend can be used to interpret the map symbols, and the scale is used to determine the relationship between distances shown on the map and actual distances. Problem of determining and tracking robot's position relative to map of environment. (Passive vs Active & Local vs Global)

closed 2020-08-26 by wohe

imstevenpmwyca, 2021-03-03:
@grieneis Did you manage to use the overlapping submaps trimmer?