cartographer-project/cartographer #1541: Question: why is gravity alignment needed for 2D SLAM without IMU data?

kings-landing, 2019-03-14:
For 2D SLAM, when there is no IMU data, gravity alignment is not identity matrix, but an in-plane rotation. Why is it? Isn't gravity alignment used for correcting pitch?
```
    // There is no IMU data until 'time', so we advance the ImuTracker and use
    // the angular velocities from poses and fake gravity to help 2D stability.
    imu_tracker->Advance(time);
    imu_tracker->AddImuLinearAccelerationObservation(Eigen::Vector3d::UnitZ());
    imu_tracker->AddImuAngularVelocityObservation(
        odometry_data_.size() < 2 ? angular_velocity_from_poses_
                                  : angular_velocity_from_odometry_);
```

gaschler, 2019-03-14:
`PoseExtrapolator` always comes up with some 3D rotation.
If there is no IMU, it looks at wheel odometry, or else at orientation changes in scan matching an extrapolates that.

My feeling is that the z-rotation of gravity alignment will "cancel out" and the scan matchers don't care about it, but I didn't check.

kings-landing, 2019-03-14:
Yeah, I think it will cancel out the z-rotation. But I don't understand why it's needed when there is no IMU data for 2D SLAM. Is it helpful in any way in this case (2D SLAM and no IMU)?

gaschler, 2019-03-15:
The current design of 2D SLAM is flexible allows all combinations from 2D laser input to multiple 3D lidars, IMU, odometry.
Deutsches Museum dataset is a good example for 3D scans in 2D SLAM, it works well.

Closing as answered.

closed 2019-03-15 by gaschler