cartographer-project/cartographer #1545: Stuck when trying to launch cartographer with RPLidar

joyde3382, 2019-03-18:
Hey i have this issue when running the command 

`roslaunch cartographer_ros backpack_2d.launch´

The process starts but gets stuck like in the picture below

[image]

I am fully able to run the deutsche museum demo, provided by Cartographer, but i can't get it to work with my own lidar (RPLidar a1).

Also the lidar is perfectly connected (i get data from /scan) and i am running hector SLAM just fine.

Any suggestions to get it working with cartographer?

carry4985, 2019-03-20:
I guess you need check the configuration file(.lua). you should set num_laser_scans = 1 and num_multi_echo_laser_scans = 0

joyde3382, 2019-03-21:
> I guess you need check the configuration file(.lua). you should set num_laser_scans = 1 and num_multi_echo_laser_scans = 0

Thanks for your reply, but I already did that, still same issue :/

here are my .lua and .launch files

backback_2d.lua:  https://pastebin.com/FXA4Du0D
backback_2d.launch:  https://pastebin.com/trzJDZ7Z

carry4985, 2019-03-22:
I set the base_frame is laser and the topic is 'scan'. Attached to the file I have changed.

joyde3382, 2019-03-25:
Hmm, not quite sure what you're getting at. I managed to link the lidar node with the cartographer node 
by replacing the following lines: 

  tracking_frame = "base_link",
  published_frame = "base_link",

with

  tracking_frame = "rplidarNode",
  published_frame = "rplidarNode",

[image]

However i am still stuck at the same place..

jgaud, 2019-11-15:
@joyde3382 Did you figure out how to set it up?
I am running into the same issue.

Thanks.