cartographer-project/cartographer #1546: Question on multiple lidar SLAM

albertchang1989, 2019-03-19:
Recently, I plan to use two velodyne lidars and an IMU to build the map. I am wondering:
1. Wheather the precision of the result will be improved or not with more than one lidar when SLAM ?
2. Does cartographer provide a way to estimate  the extrinsics parameter between two lidars?

Thank you for any help you can offer !

D4YDR34MPH4N70M, 2019-04-10:
direction and distance because of unequal spacing of parallels, but great circles form straight lines, so plotting air travel routes is easier. When conic projections are fitted together, they produce a continuous map and preserve the relative size and shape of small areas.

D4YDR34MPH4N70M, 2019-04-10:
the scale remains the same with any system of measurement.