cartographer-project/cartographer #1561: submaps built with cartographer are blur

yangtfu, 2019-04-05:
Hi. everyone. I was building a map of my office with both cartographer and gmapping. We have a 16 channel lidar on top of a robot providing both scan and point cloud data and a single channel lidar mounted at waist level of the robot providing only scan data. However the maps built with cartographer are always not satisfactory, the walls are thick and corners are not vertical compared with the one built with gmapping. I thought the submaps are poor and tuned the parameters related to ceres matcher and realtime correlative scan matcher, but the results are still not very good. I uploaded the bags and maps below, could anyone please check them for me. Thanks a lot.
The map built with gmapping
[https://drive.google.com/open?id=1lzsTSdhI5qMcxPly_Wlq8mBNgbmuYIU4](url)
The map built using scan data with cartographer 
[https://drive.google.com/open?id=12IwNWz8H7NQIhJQEjXIxpEBpMqNZXA2T](url)
The map built using point cloud data with cartographer
[https://drive.google.com/open?id=1vNNvQaa_h7zVUjAGS0UE0jVfhWV4MU4-](url)
Bag file:
[https://drive.google.com/open?id=1KKTtZpR1W78w_S2_IsPZE0DybuWN2AmY](url)
lua file for scan data builder:
```lua
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "base_footprint",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 0.5,
  odometry_sampling_ratio = 0.5,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.max_range = 30.
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 30
```
lua file for point cloud builder:
```lua
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "base_footprint",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 0.5,
  odometry_sampling_ratio = 0.5,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}
MAP_BUILDER.use_trajectory_builder_2d = true
--MAP_BUILDER.num_background_threads = 7

TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.min_range = 0.4
TRAJECTORY_BUILDER_2D.max_range = 50.
TRAJECTORY_BUILDER_2D.min_z = 0
TRAJECTORY_BUILDER_2D.max_z = 2.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 5.
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.voxel_filter_size = 0.025

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.2
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(20.)
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 1e-1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 100
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 400
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.ceres_solver_options.use_nonmonotonic_steps = true
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.ceres_solver_options.max_num_iterations = 40
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.ceres_solver_options.num_threads = 4

TRAJECTORY_BUILDER_2D.submaps.grid_options_2d.resolution = 0.05
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 30

--TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1

POSE_GRAPH.optimize_every_n_nodes = 10

--POSE_GRAPH.constraint_builder.sampling_ratio = 0.3
--POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.constraint_builder.min_score = 0.1
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.6
```

qust001, 2019-04-16:
follow

zxtsuper, 2019-04-26:
flow

zxtsuper, 2019-04-26:
follow

qintxwd, 2020-05-21:
follow