cartographer-project/cartographer #1565: Question: what's the difference between node and submap?

kings-landing, 2019-04-11:


gaschler, 2019-04-11:
In most cases, every laser scan rotation becomes one node.
If there are multiple lidars, most people set the parameters that one rotation time span goes into one node.
The data inside a node is a (filtered) point cloud.

A submap is a grid structure (2D or 3D) of occupancy values.
It is built over 20-200 nodes or laser rotations.

> Looks like every successful insert to submap becomes node? Isn't this making pose graph optimization problem too big?

The pose graph is big, but optimization is not much effort for hundreds of submaps. Its structure is mostly a trajectory with sparse loop closure constraints.

JD-ETH, 2019-04-14:
@gaschler Are you sure every successful scan becomes one node? I have been running cartographer with Ouster at 10 Hz, but the resulting trajectory in protobuf-stream file only has 2 nodes every second. Is there subsampling happening somewhere?

kings-landing, 2019-05-09:
@gaschler Thanks a lot for your answer!
I have a follow-up question. In pose graph optimization, it looks like node poses get adjusted / refined, but submap probability grids are not re-generated according to their internal nodes' pose adjustment. Why is it?

gaschler, 2019-05-10:
> submap probability grids are not re-generated according to their internal nodes' pose adjustment

Correct, submaps are only moved during optimization, but their probability grid is not changed (or distorted).

> Why is it?

Recomputing the grid in a submap is quite an effort, so it would make optimization orders of magnitude slower. I'm not aware of such kind of experiment in Cartographer.

That aside, if drift is high compared to the size of a submap, you can choose smaller submap sizes.

ojura, 2019-05-10:
@gaschler ironically, you can actually increase the drift that way, because local slam has less features to work with.

Btw, if you allow the optimiser to optimise all trajectory nodes, you can sometimes end up with the following:
[image]

The constraints pull in their corresponding trajectory nodes a bit stronger than the rest of the nodes in that submap. Yes, the answer might be a higher constraint builder sampling ratio, but that can end up being quite computationally expensive with larger submaps and finer resolutions.

Maybe we could optimise just the submap poses instead of using intra constraints to tie nodes to submaps?

kings-landing, 2019-05-10:
@gaschler Thanks a lot for your answer!
@ojura you asked my question :) 
If submap is not recomputed, why not only optimize submap poses?

ojura, 2019-05-10:
Yes, I phrased it a bit different :) It's not straightforward, though - you would have to move the nodes with submaps, and choose a way to assign each node to a submap (perhaps assign it to the older submap in the active submap pair). We would have to avoid sudden jumps between nodes which belong to adjacent submaps, and probably have to smooth the node poses according to the submaps they belong to.

closed 2020-07-07 by wohe