cartographer-project/cartographer #1567: Submap of outdoor scenes

Yuhaizhen, 2019-04-12:
The Submap is not continuous and the trajectory is not accurate enough when the cartographer 3D is run in real time at a speed of 5 km/h In the outdoor scene。
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gaschler, 2019-04-12:
Please follow cartographer_ros's issue template and post the text output of rosbag_validate and your lua and urdf files.

chenxianbo, 2019-07-04:
you can watch the cpu

closed 2020-06-25 by wohe