cartographer-project/cartographer #1569: Error using Odometry data in 2d Cartographer

shrijan00, 2019-04-15:
Hi all,

So I recorded a bag file consisting of 2d lidar scan, tf, tf_static and odometry data(fusing imu and wheel odom).
`rosbag record /tf /tf_static /odometry/filtered /scan`

So when i try to use the bag file in cartography without the odometry, it works perfectly fine.
In that case I'm using this configuration

`include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.15
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(35.)

POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options`
and my launch file is this 
`
<launch>
  <param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename pr2.lua"
      output="screen">
    <remap from="scan" to="/scan" />
    <remap from="odom" to="/odometry/filtered" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)">
    <remap from="tf" to="tf_in" />
  </node>
</launch>`
and my tf tree is proper in this case 
[image]

So, the issue comes when I want to use the odometry. For that I just change 
`use_odometry = true` in the above lua. And using the same launch file when I try to run I get this error
[image]
 Can anyone please help me how to enable the odometry so that the cartographer can access both lidar and odometry data to build map and localise itself.

qust001, 2019-04-16:
the same to me

shrijan00, 2019-04-16:
So, 
I recorded my bag once again including the /clock as well this time. I ran the same configuration by setting use_odometry = true and it is working now.
But I'm getting warning messages as you can see in the image.
[image]

I guess it must be something related to timestamp at which the sensors data are recorded. Please correct me if I'm wrong.

wilcobonestroo, 2019-04-16:
What command do you use to play back the rosbag? (do you use --clock or not?)

shrijan00, 2019-04-17:
Yes, I do use --clock.
You can see my node playbag (3rd line from the bottom) in the launch file i posted earlier.

MichaelGrupp, 2019-04-17:
1. Check if your sensor data is strictly time ordered. You can use the `cartographer_rosbag_validate` tool for that as described [in the documentation of Cartographer ROS](https://google-cartographer-ros.readthedocs.io/en/latest/your_bag.html?highlight=validate#validate-your-bag). You wrote that the error is gone by decreasing the sampling ratio - this could suggest that you just sampled out an invalid data chunk.

2. The second warning could be caused by replaying the bagfile in the same ROS core session, i.e. resetting the published `--clock` and affecting some node. If you don't find the root of this, restart the `roscore` and other nodes you might have together with your Cartographer session.
For debugging purposes you can also have a look at `cartographer_dev_rosbag_publisher` instead of simulation time, which republishes the data with fresh timestamps and can be called repeatedly without annoying warnings.

wilcobonestroo, 2019-04-17:
I assume you have recorded the clock as well. If you use `--clock` there will be multiple publishers on the clock topic. Try running without the `--clock` option.

shrijan00, 2019-04-22:
Hi, 
Sorry the for the delayed response
@MichaelGrupp @wilcobonestroo   I did the Validation this time as mentioned. Here is the screenshot for the bag file.
[image]
So I followed like what you suggested that there must be multiple publishers on the clock topic, so I removed --clock from my launch file.
` <node name="playbag" pkg="rosbag" type="play"
      args="$(find cartographer_ros)/bag/new_test.bag">`

And indeed it was the issue and so no more warning messages. So everything looks good. The only thing is in order to run the Cartographer I'm using a low value for odometry_sampling_ratio as mentioned earlier.

ojura, 2019-05-11:
@shrijan00 Low value for the odometry sampling ratio is only hiding the problem. That is a bad solution.

Guys, sorry for introducing these sampling ratio options, it is partially my fault. I originally introduced it for the lidar (which actually makes a lot of sense, it’s a cheap way to speed things up if slam works okay for you using a fraction of the lidar data).

Then, for “completeness” we added the  sampling ratios for odometry and also for the imu (which is especially silly, in retrospective). The result is that many people concluded they could “fix” their timing issues by just decimating their odometry or imu data. I propose we remove the odometry and imu sampling ratio options, or print fat warning or require special flags to discourage users from using these options.

shrijan00, 2019-05-17:
@ojura Yeah Exactly. Even I was not satisfied with the following solution but it made it work so I ignored it. Apart from using odometry sampling ratio, there is no other way to fix this?

Fanny-one, 2019-10-09:
@shrijan00 
Hi, I have met the same problem and the above replys give me a good answer.
Also , I find you are testing the same project very likely. Now what about your progress and can we  
communicate with each other?