cartographer-project/cartographer #1573: Docs / Examples on using standalone Cartographer (without ROS)

nrasulnrasul, 2019-04-30:
Hi,
I am planning to use Standalone Cartographer (i.e without ROS ). 
I have gone through the documentation [cartographer.readthedocs](https://google-cartographer.readthedocs.io/en/latest/) but it was not enough to start feeding the live sensors data( not from rosbag ) and running the SLAM.

I have found this [similar issue 1229](https://github.com/googlecartographer/cartographer/issues/1229) but it was solved by running offline node with data from rosbag files.

I would appreciate any examples/ tutorials/ API documentation on setting up and using the standalone version with live data.

Thank you.

alankdkd, 2019-06-20:
I'm looking for the same information.  I'm sure it exists somewhere, but I haven't been able to find it.  (apache8080 is also looking for it: https://github.com/googlecartographer/cartographer/issues/1584)

dmitry-akimov-dXgjvQjNiMuVYecJ, 2019-06-21:
This project is essentially undocumented. Only the ROS component has _some_ documentation.

alankdkd, 2019-06-21:
Doh!  I suspected that was the case.  Maybe us users can pool our information somehow on github.

ojura, 2019-06-21:
The ros component can serve as a nice template for using the library, though. A lot of code in there can be reused, that was my motivation behind ros_stub.

alankdkd, 2019-06-21:
Ojura, can your ros_stub be used as a starting point for development?  Can you tell me more about it?

Another topic: I'm trying to figure out how to get data from different sources into Cartographer.  So far I've encountered Protobuf3, .bag files, messages and topics.  It's confusing.

Can anyone clear up how Cartographer gets its inputs, and where the formats are defined?  Thanks!

RicheyHuang, 2019-07-10:
@nrasulnrasul Hi, this can be achieved by referring the test files(map_builder_test.cc) in source code. I just ran the whole SLAM process using my own data from a live camera by the cartographer standalone library. It works for sure :)

nrasulnrasul, 2019-07-10:
@RicheyHuang  Thanks for the reply.  I have referred the cartographer ros package and I am able to run cartographer. Documentation of APIs would have been more useful instead of everyone start doing reverse engineering to understand the functions and flow.

closed 2020-05-29 by wohe