cartographer-project/cartographer #1578: Race condition on quitting - global sam uses probability tables alloc'd in local SLAM

ojura, 2019-05-20:
I think there is this a condition which can happen when quitting Cartographer. Below is a stack trace produced by Valgrind.

It looks like `FastCorrelativeScanMatcher2D` in global slam is using probability lookup tables from `ActiveSubmaps2D`. `Submap2D` contains a plain pointer to `ValueConversionTables`. During destruction, we can destroy `LocalTrajectoryBuilder2D` while the worker pool in global slam is still looking for constraints.

```
I0520 17:25:27.247808   651 map_builder_bridge.cc:213] Finishing trajectory with ID '0'...
I0520 17:25:27.285739   654 pose_graph_2d.cc:642] Remaining work items in queue: 0
Optimizing: 0.0%...==474== Thread 15:
==474== Invalid read of size 8
==474==    at 0x459B68: GetCorrespondenceCost (grid_2d.h:62)
==474==    by 0x459B68: GetValue (occupied_space_cost_function_2d.cc:80)
==474==    by 0x459B68: Evaluate (cubic_interpolation.h:293)
==474==    by 0x459B68: Evaluate<ceres::Jet<double, 3> > (cubic_interpolation.h:344)
==474==    by 0x459B68: operator()<ceres::Jet<double, 3> > (occupied_space_cost_function_2d.cc:56)
==474==    by 0x459B68: Call (variadic_evaluate.h:176)
==474==    by 0x459B68: ceres::internal::AutoDiff<cartographer::mapping::scan_matching::(anonymous namespace)::OccupiedSpaceCostFunction2D, double, 3, 0, 0, 0, 0, 0, 0, 0, 0, 0>::Differentiate(cartographer::mapping::scan_matching::(anonymous namespace)::OccupiedSpaceCostFunction2D const&, double const* const*, int, double*, double**) (autodiff.h:289)
==474==    by 0x7241784: ceres::internal::ResidualBlock::Evaluate(bool, double*, double*, double**, double*) const (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x71D7430: ??? (in /usr/lib/libceres.so.1.13.0)
==474==    by 0xF982ECE: GOMP_parallel (in /usr/lib/x86_64-linux-gnu/libgomp.so.1.0.0)
==474==    by 0x71D957A: ceres::internal::ProgramEvaluator<ceres::internal::ScratchEvaluatePreparer, ceres::internal::DenseJacobianWriter, ceres::internal::NullJacobianFinalizer>::Evaluate(ceres::internal::Evaluator::EvaluateOptions const&, double const*, double*, double*, double*, ceres::internal::SparseMatrix*) (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x7260D7D: ceres::internal::TrustRegionMinimizer::EvaluateGradientAndJacobian() (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x726180C: ceres::internal::TrustRegionMinimizer::IterationZero() (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x7265F63: ceres::internal::TrustRegionMinimizer::Minimize(ceres::internal::Minimizer::Options const&, double*, ceres::Solver::Summary*) (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x7257799: ceres::Solver::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*) (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x7258548: ceres::Solve(ceres::Solver::Options const&, ceres::Problem*, ceres::Solver::Summary*) (in /usr/lib/libceres.so.1.13.0)
==474==    by 0x455B01: cartographer::mapping::scan_matching::CeresScanMatcher2D::Match(Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, cartographer::transform::Rigid2<double> const&, std::vector<cartographer::sensor::RangefinderPoint, std::allocator<cartographer::sensor::RangefinderPoint> > const&, cartographer::mapping::Grid2D const&, cartographer::transform::Rigid2<double>*, ceres::Solver::Summary*) const (ceres_scan_matcher_2d.cc:103)
==474==    by 0x4888D2: cartographer::mapping::constraints::ConstraintBuilder2D::ComputeConstraint(cartographer::mapping::SubmapId const&, cartographer::mapping::Submap2D const*, cartographer::mapping::NodeId const&, bool, cartographer::mapping::TrajectoryNode::Data const*, cartographer::transform::Rigid2<double> const&, cartographer::mapping::constraints::ConstraintBuilder2D::SubmapScanMatcher const&, std::unique_ptr<cartographer::mapping::PoseGraphInterface::Constraint, std::default_delete<cartographer::mapping::PoseGraphInterface::Constraint> >*) (constraint_builder_2d.cc:242)
==474==  Address 0x1b7c3d80 is 0 bytes inside a block of size 24 free'd
==474==    at 0x4C3123B: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==474==    by 0x43809C: operator() (unique_ptr.h:78)
==474==    by 0x43809C: ~unique_ptr (unique_ptr.h:268)
==474==    by 0x43809C: ~pair (stl_pair.h:208)
==474==    by 0x43809C: destroy<std::pair<const std::tuple<float, float, float>, std::unique_ptr<const std::vector<float> > > > (new_allocator.h:140)
==474==    by 0x43809C: destroy<std::pair<const std::tuple<float, float, float>, std::unique_ptr<const std::vector<float> > > > (alloc_traits.h:487)
==474==    by 0x43809C: _M_destroy_node (stl_tree.h:650)
==474==    by 0x43809C: _M_drop_node (stl_tree.h:658)
==474==    by 0x43809C: _M_erase (stl_tree.h:1858)
==474==    by 0x43809C: ~_Rb_tree (stl_tree.h:949)
==474==    by 0x43809C: ~map (stl_map.h:294)
==474==    by 0x43809C: ~ValueConversionTables (value_conversion_tables.h:32)
==474==    by 0x43809C: ~ActiveSubmaps2D (submap_2d.h:79)
==474==    by 0x43809C: cartographer::mapping::LocalTrajectoryBuilder2D::~LocalTrajectoryBuilder2D() (local_trajectory_builder_2d.cc:49)
==474==    by 0x495F7A: operator() (unique_ptr.h:78)
==474==    by 0x495F7A: ~unique_ptr (unique_ptr.h:268)
==474==    by 0x495F7A: ~GlobalTrajectoryBuilder (global_trajectory_builder.cc:47)
==474==    by 0x495F7A: cartographer::mapping::(anonymous namespace)::GlobalTrajectoryBuilder<cartographer::mapping::LocalTrajectoryBuilder2D, cartographer::mapping::PoseGraph2D>::~GlobalTrajectoryBuilder() (global_trajectory_builder.cc:47)
==474==    by 0x3AA934: operator() (unique_ptr.h:78)
==474==    by 0x3AA934: ~unique_ptr (unique_ptr.h:268)
==474==    by 0x3AA934: ~CollatedTrajectoryBuilder (collated_trajectory_builder.h:48)
==474==    by 0x3AA934: cartographer::mapping::CollatedTrajectoryBuilder::~CollatedTrajectoryBuilder() (collated_trajectory_builder.h:48)
==474==    by 0x2EFA0D: operator() (unique_ptr.h:78)
==474==    by 0x2EFA0D: ~unique_ptr (unique_ptr.h:268)
==474==    by 0x2EFA0D: _Destroy<std::unique_ptr<cartographer::mapping::TrajectoryBuilderInterface> > (stl_construct.h:98)
==474==    by 0x2EFA0D: __destroy<std::unique_ptr<cartographer::mapping::TrajectoryBuilderInterface>*> (stl_construct.h:108)
==474==    by 0x2EFA0D: _Destroy<std::unique_ptr<cartographer::mapping::TrajectoryBuilderInterface>*> (stl_construct.h:137)
==474==    by 0x2EFA0D: _Destroy<std::unique_ptr<cartographer::mapping::TrajectoryBuilderInterface>*, std::unique_ptr<cartographer::mapping::TrajectoryBuilderInterface> > (stl_construct.h:206)
==474==    by 0x2EFA0D: ~vector (stl_vector.h:434)
==474==    by 0x2EFA0D: ~MapBuilder (map_builder.h:40)
==474==    by 0x2EFA0D: cartographer::mapping::MapBuilder::~MapBuilder() (map_builder.h:40)
==474==    by 0x33ACCE: operator() (unique_ptr.h:78)
==474==    by 0x33ACCE: ~unique_ptr (unique_ptr.h:268)
==474==    by 0x33ACCE: cartographer_ros::MapBuilderBridge::~MapBuilderBridge() (map_builder_bridge.h:59)
==474==    by 0x319BD0: cartographer_ros::Node::~Node() (node.cc:171)
==474==    by 0x2D5654: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
==474==    by 0x2F3791: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
==474==    by 0x2F3791: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
==474==    by 0x6A05DB2: QThreadPrivate::start(void*) (in ../Qt/5.12.2/gcc_64/lib/libQt5Core.so.5.12.2)
==474==    by 0x836F6DA: start_thread (pthread_create.c:463)
==474==    by 0xB0F088E: clone (clone.S:95)
==474==  Block was alloc'd at
==474==    at 0x4C3017F: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==474==    by 0x3B61E9: make_unique<std::vector<float> > (memory.h:173)
==474==    by 0x3B61E9: PrecomputeValueToBoundedFloat (value_conversion_tables.cc:42)
==474==    by 0x3B61E9: cartographer::mapping::ValueConversionTables::GetConversionTable(float, float, float) (value_conversion_tables.cc:61)
==474==    by 0x393867: cartographer::mapping::Grid2D::Grid2D(cartographer::mapping::MapLimits const&, float, float, cartographer::mapping::ValueConversionTables*) (grid_2d.cc:69)
==474==    by 0x394930: cartographer::mapping::ProbabilityGrid::ProbabilityGrid(cartographer::mapping::MapLimits const&, cartographer::mapping::ValueConversionTables*) (probability_grid.cc:31)
==474==    by 0x397731: make_unique<cartographer::mapping::ProbabilityGrid, cartographer::mapping::MapLimits, cartographer::mapping::ValueConversionTables*> (memory.h:173)
==474==    by 0x397731: cartographer::mapping::ActiveSubmaps2D::CreateGrid(Eigen::Matrix<float, 2, 1, 0, 2, 1> const&) (submap_2d.cc:198)
[... cut ...]
==474==    by 0x4979A6: cartographer::mapping::(anonymous namespace)::GlobalTrajectoryBuilder<cartographer::mapping::LocalTrajectoryBuilder2D, cartographer::mapping::PoseGraph2D>::AddSensorData(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cartographer::sensor::TimedPointCloudData const&) (global_trajectory_builder.cc:58)
==474==    by 0x3A8FAC: cartographer::sensor::Dispatchable<cartographer::sensor::TimedPointCloudData>::AddToTrajectoryBuilder(cartographer::mapping::TrajectoryBuilderInterface*) (dispatchable.h:35)
==474== 
==474== (action on error) vgdb me ... 
```

ojura, 2019-05-20:
Related: #1446

ajakhotia, 2019-05-24:
I got bit by this problem as well. I think the race condition mentioned by @ojura is exposing the underlying problem which is that `Submap2D` and `Grid2D` are holding a raw pointers to, technically, a shared resource(`ValueConversionTables`). Moving the `ValueConversionTables` to `MapBuilder` will definitely suppress this life-time issue but the underlying problem will continue to exist and will prop-up again whenever users(like me) try to use `Submap2D` for scan matching in contexts outside of the current cartographer stack. This combined with with the fact that `Submap2D`'s are communicated to the rest of the process using `shared_ptr<>` makes it worse furthermore.

That said, I have a different proposal than what @ojura mentioned but it shares the motivation:
**Modify `ActiveSubmaps`, `Submap2D`, and `Grid2D` to use `std::shared_ptr<ValueConversionTables>`**.
This will ensure that the conversion tables will live at least as long as the last `Submap2D` or `Grid2D` that refers to them.

This  problem is currently blocking me. If the authors with bigger context than myself think it's a sane idea, I can start working on a PR to address this.

Also, related to this but tangential:
Both `Submap2D` and `Grid2D` hold a pointer to the `ValueConversionTables`. Technically, `Submap2D` has no real purpose for holding  this pointer as the member `Grid2D` already has it. If others here agree that I am not being naive about it, I can go ahead an create a second PR to **Remove redundant references to the `ValueConversionTables` from `Submap2D`**.

ojura, 2019-05-25:
A shared pointer sounds like a reasonable idea as well. I also thought of it, but initially suggested the lighter change first because perhaps the intention behind using a plain pointer was for performance reasons.