cartographer-project/cartographer #1579: map_builder.lua was not found

cmoxie, 2019-05-22:
Hi there, 

We were running our cartographer ROS node that has been up to this point working well. We received the error message: '[FATAL] [1558546404.710037245]: F0522 11:33:24.000000 29702 configuration_file_resolver.cc:45] File 'map_builder.lua' was not found.'

map_builder.lua does exist in install_isolated/share/cartographer/configuration_files

Our configuration file is a lightly modified version of the backpack_2d.lua example file. 

```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link", --os1_imu,base_link
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 2.0,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.min_range = 1
TRAJECTORY_BUILDER_2D.min_z = 0
TRAJECTORY_BUILDER_2D.imu_gravity_time_constant = 1
POSE_GRAPH.optimization_problem.log_solver_summary = false

return options
```

We are on the most current branch of cartographer. Also if we reference backpack_2d.lua it also does not find the map_builder.lua file

'<launch>

        <node pkg="rosserial_python" type="serial_node.py" name="serial_node">
                <param name="port" type="string" value="/dev/ttyACM0"/>
                <param name="buad" type="int" value="57600"/>
        </node>

	<node pkg="tf2_ros" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0.16 0 0 0 1 base_link os1" />
	<node pkg="tf2_ros" type="static_transform_publisher" name="link2_broadcaster" args="0 0 0.16 0 0 0 1 base_link os1_imu"  />

	<node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">
           <remap from="cloud_in" to="/os1_node/points" />
              <rosparam>
                   target_frame: os1
                   transform_tolerance: 0.01
                   min_height: 0.0
                   max_height: 1.0
                   angle_min: -3.14 
                   angle_max: 3.14
                   angle_increment: 0.0087
                   scan_time: 0.3333
                   range_min: 0.45
                   range_max: 10.0
                   user_inf: true
               </rosparam>
         </node>

        <node name="cartographer_node" pkg="cartographer_ros"
           type="cartographer_node" args="
           -configuration_directory $(find cartographer_ros)/configuration_files
           -configuration_basename backpack_2d.lua"
           output="screen">
	   <remap from="points2" to="/os1_node/points" />
           <remap from="imu" to="/os1_node/imu" />
        </node>

        <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
           type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

	<node pkg="teleoperation" type="tfToPose" name="tfToPose" output="screen"/>

        <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
           <rosparam file="$(find zeb_2dnav)/costmap_common_params.yaml" command="load" ns="global_costmap" />
           <rosparam file="$(find zeb_2dnav)/costmap_common_params.yaml" command="load" ns="local_costmap" />
           <rosparam file="$(find zeb_2dnav)/local_costmap_params.yaml" command="load" />
           <rosparam file="$(find zeb_2dnav)/global_costmap_params.yaml" command="load" />
           <rosparam file="$(find zeb_2dnav)/base_local_planner_params.yaml" command="load" />
           <rosparam>
              planner_frequency: 0.5
           </rosparam>
        </node>
        
        <node pkg="teleoperation" type="motorController" name="motorController" output="screen" />

</launch>`

If you have any suggestions to what we did wrong that would be appreciated

Thanks

Ruthrash, 2019-07-02:
Hey, were you able to solve this problem? I am stuck with the same.

RSFalcon7, 2019-07-15:
I had a similar issue.
Delete the `build`, `devel` and `install` directories, build it again and source the `install_isolated/setup` solved

qmx066, 2019-08-27:
i'm having the same issue, all configuration files are good in install folder, source install/setup.bash works fine running executable but doesn't help finding map_builder.lua...
appreciate if anyone helps.

charlotteheggem, 2019-12-04:
Did you solve this? I am having the same problem..

szybalint, 2020-02-21:
I just copied the files from install_isolated/share/cartographer/configuration_files to install_isolated/share/cartographer_ros/configuration_files and everything works fine. Had the same issue before I did that.

echoand, 2022-10-19:
put it to install_isolated/share/cartographer_ros/configuration_files instead of  install_isolated/share/cartographer/configuration_files