cartographer-project/cartographer #1582: Barometer support?

lukehutch, 2019-05-24:
I need to increase the absolute altitude accuracy of long scan paths by means of a barometer. I have a MS5611 barometer chip I can get readings from.

In [this comment](https://github.com/googlecartographer/cartographer/issues/1123#issuecomment-384920388), @ojura said "I believe the team at some point used a barometer for measuring height".

How can I incorporate barometer readings into Cartographer?

ojura, 2019-05-27:
The original author of Cartographer quickly corrected me that I was mistaken. Indeed different height submaps are displayed on the GIF on the repo frontpage, but the height there is from 3D SLAM, not from an external height reading such as a barometer.

I would point you to the pose graph. It's a quite flexible tool. You could, for example, constrain the height of the trajectory nodes according to your barometer readings with some strength which will gently pull the trajectory towards your altitude readings, but not too hard (otherwise you will start trusting the noise from your barometer).

lukehutch, 2019-05-27:
Thanks, I'll try that. How do I reliably synchronize timestamps when adding a new sensor like this?

wolfgangschwab, 2021-07-09:
@lukehutch, 
have you found a solution for integrating a barometer into cartographer?

lukehutch, 2021-07-09:
> @lukehutch,
> have you found a solution for integrating a barometer into cartographer?

No, I wrote my own SLAM software from scratch after failing to get Cartographer to do what I needed (not just in terms of integrating the barometer signal, but also Cartographer had major drift issues with my datasets, and I couldn't seem to get it to behave properly...).

wolfgangschwab, 2021-07-09:
OK, interesting solution :-o. Does not sound like a quick and easy solution. I have respect for writing an own SLAM Software. Is your SLAM Software publicly available?

lukehutch, 2021-07-09:
Writing a new point cloud SLAM system is insanity, I don't recommend it :-D  It's a great learning opportunity though! 

It works well for many uses, but there are a couple of usecases that still need improvement. It's not available yet, unfortunately, and it's complex enough that I wouldn't want to support it for anything other than my own project. It's specifically tailored to work with an Ouster OS1-64, and would need some nontrivial work done to port to another sensor. Sorry about that!

wolfgangschwab, 2021-07-09:
Thanks. So no solution for my problem. But I wish you good luck for solving the outstanding issues and for the future. :-)

lukehutch, 2021-07-09:
Thanks! If the software ever becomes mature, I'll see if I can release it, but that's quite a long way off, I think. I hope you get your barometer signal integrated -- it's a very important signal!

wolfgangschwab, 2021-07-20:
@lukehutch , I added the usage of barometer and compass to cartographer. I'm not exactly sure whether it works correctly. Do you know how I can check in the logs how the pose is calculated? If you are interested in the solution send me an pm.

lukehutch, 2021-07-20:
@wolfgangschwab sorry, I don't know enough about Cartographer!