cartographer-project/cartographer #1590: Delete Trajectories from MapBuilder

pHaeusler, 2019-06-11:
Thanks for all of the hard work on this library!

I have a use case where I would like to remove a trajectory used by a robot for localization (there is a `PureLocalizationTrimmer` on this trajectory). In future the robot may need to start localizing again, so the trajectory would be re-added. There are also a set of consistent frozen trajectories used for localization.

I'm trying to delete this trajectory and its associated `TrajectoryBuilderInterface` from the `MapBuilder`.

For reference: [DeleteTrajectory #1205](https://github.com/googlecartographer/rfcs/blob/master/text/0023-delete-load.md)

It looks like `PoseGraph2d` and `PoseGraph3d` don't actually remove trajectories marked for deletion in `DeleteTrajectoriesIfNeeded()`, they just remove the associated submaps from the pose graph and mark the trajectory as deleted.

It also doesn't look like there is an API to remove a trajectory builder, is that right?

Wouldn't this cause a long term growth in previously deleted trajectories? Is this the desired behavior?

joehrobot, 2019-07-17:
Hi thanks for this awesome project!

I have the same problem like above issue.
any comments would be appreciated.

PeakDing, 2020-10-23:
Hi thanks for this awesome project! I have the same question.
I used the grpc deleteTrajectory API but it seems dont't work,
any comments would be appreciated.

PeakDing, 2020-10-28:
maybe I find the answer,
first, you need to save the .pbstream file with include_unfinished_submaps: true
second,  load the .pbstream file with  load_frozen_state: fasle
next, you need to finish the trajectory firstly and then delete it.
I think the reason is that some nodes which belong to the last unfinished_submap are omitted to be deleted

qetuo105487900, 2024-10-08:
anyone know the detail work?