cartographer-project/cartographer #1592: Example maps?

v4hn, 2019-06-17:
Hello there,

I am currently looking for a set of open 2d example maps people generated with Cartographer in different environments and only find slides or pictures without meta data.
I aim to use them as a basis for simulation experiments.
Is anyone aware of a resource of such maps?

If you find this is inappropriate as an issue here, feel free to close it unanswered.

wohe, 2020-06-25:
Not sure I understand the question. Why would you not simply compute some maps from the public data?
Public data: https://google-cartographer-ros.readthedocs.io/en/latest/data.html
Demo: https://google-cartographer-ros.readthedocs.io/en/latest/demos.html

closed 2020-06-25 by wohe

v4hn, 2020-06-25:
Thanks! I could have made good use of that hint a year ago and it fully resolves the issue! :sun_behind_small_cloud: 
You are of course right, looking at raw runs makes so much more sense than collecting results...