cartographer-project/cartographer #1593: Tuning config params of standalone Cartographer

nrasulnrasul, 2019-06-17:
I am using standalone cartographer without ROS, and below is the map generated so far with the option provided below.

[image]

I suppose that local slam params are tuned well as I can see the submaps but the problem is with posegraph parameters since submaps are not getting matched.

Any suggestion on which parameters I should tweak to the submaps properly.

Thanks.

```lua
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
}

MAP_BUILDER.use_trajectory_builder_2d = true
MAP_BUILDER.num_background_threads = 7

-- INPUT
TRAJECTORY_BUILDER_2D.min_range = 0.2
TRAJECTORY_BUILDER_2D.max_range = 20.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 0
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.max_range = 20.0
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.max_range = 20.0
TRAJECTORY_BUILDER_2D.loop_closure_adaptive_voxel_filter.max_range = 20.0

-- LOCAL SLAM

TRAJECTORY_BUILDER_2D.submaps.num_range_data = 100

--------------CeresScanMatcher
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.occupied_space_weight = 1
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 10
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 40

--------------RealTimeCorrelativeScanMatcher
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(10.)
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

-- Global SLAM
POSE_GRAPH.optimize_every_n_nodes = 30
POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.log_matches = true
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window = 15.0
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window = math.rad(30)
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 200s
POSE_GRAPH.constraint_builder.sampling_ratio = 0.01

POSE_GRAPH.optimization_problem.log_solver_summary = true
POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options
```

closed 2019-06-24 by nrasulnrasul

lyeemax, 2019-06-26:
how do you do that without ros?Amazing!could you please share some experiences on that?really appreciate

nrasulnrasul, 2019-06-27:
I went through `cartographer_ros` package to get how `cartographer` apis were used. It took a while but it is not difficult to use the cartographer library and so far I tried 2d mapping and the resulting maps were good. In which middleware are you looking to use cartographer?

lyeemax, 2019-06-27:
My aim is building a slam robot without ros as my college research,I have already built the mobile platfrom,which could get accurate localization.Now what I am going to do is to use the cartographer to implement building map and then navigate.If you are not uninterested in my humble work,I would like to invite you as collaborators

ojura, 2019-06-28:
My approach was using a ROS stub to avoid reimplementing cartographer_ros, see https://github.com/googlecartographer/cartographer/issues/1229