cartographer-project/cartographer #1600: tuning 3D cartographer slam in my own bag with VLP -16

aidamQin, 2019-06-26:
@SirVer Hi,SirVer, I try to make `  POSE_GRAPH.optimize_every_n_nodes = 0 ` to close global slam and see my local , I got the map as following as.
[image]
[image]
when I opened the global slam ,  the result was not getting better, and it was even worse ， so I wandering where the problem is and how I should turn to get better.
[Bag](https://www.dropbox.com/s/xlzrvk5mrianw0e/2019-06-17-15-41-28.bag.zip?dl=0)
[configuration](https://www.dropbox.com/s/vkxgnl66mw0owx4/configuration.zip?dl=0)
ps:I receive the laser and Imu data from the bag through a new node and republish the same  topic name and frame_id in the original demo 3D configuration file to the Cartographer.
[new Node](https://www.dropbox.com/s/0i5qxxqdfdf21iu/transformation.zip?dl=0)
Let me know if you need any other files in order to reproduce the issue. I am look forward to your reply and thank you for your attention.Wish you a nice day!

aidamQin, 2019-06-26:
@wjwwood @SirVer @damienrg @damonkohler

QiaoShin, 2019-06-28:
@ojura