cartographer-project/cartographer #1603: Why pure localization mode consumes more CPU resources than SLAM mode?

huangshuqing, 2019-07-03:
Why pure localization mode consumes more CPU resources than SLAM mode?

Yu-Xiaoxian, 2019-08-10:
In mapping, cartographer uses fast correlative scan matcher(FCS) to find loop closure, search window of  FCS is defined in pose_graph.lua, by default is 7m×7m×30°. While in pure localization, the search window becomes very large, as large as the search window. As a result, localization consumes more CPU resources. 

But there are still some methods to decrease the CPU load by tuning the parameters, you can look up the documents [Tuning methodology](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency).

huangshuqing, 2019-08-14:
hello. I am sorry for replying the email so lately. Are you chinese? can we chat with each other by chinese?

Yu-Xiaoxian, 2019-08-19:
可以没问题，你可以看constraint_builder_2d.cc中ComputeConstraint()函数的定义，在pure_localization模式下，会频繁地调用MaybeAddGlobalConstraint()，这个函数会搜索整张子图，所以造成的开销特别大。建议将global_localization_ratio调小，这样就可以从一定程度上减少开销。

以上都是凭记忆打出来的，可能跟实际函数名、参数名有出入，希望能够帮到你，如果有问题我们可以继续讨论。

huangshuqing, 2019-08-19:
是的，非常谢谢你，你说的是对的，这个问题我前一阵子看代码的时候正好看到了，后续再继续深入下去，我们能留个联系方式吗？我qq:690064483

KleinYuan, 2019-08-19:
@Yu-Xiaoxian do you mean `global_sampling_ratio ` insteaad?

Yu-Xiaoxian, 2019-08-20:
> @Yu-Xiaoxian do you mean `global_sampling_ratio ` insteaad?

Yeah, you are right. Decrease `global_sampling_ratio` and increase `global_constraint_search_after_n_seconds` will decrease the compute load efficiently.

artemiialessandrini, 2019-10-24:
lol, very informative

closed 2020-08-26 by wohe