cartographer-project/cartographer #1607: why gravity_vector_ = rotation.conjugate() * gravity_vector_ use conjugate？

tinytinywell, 2019-07-05:
Hello I have three questions,first one is the title.
1. why gravity_vector_ = rotation.conjugate() * gravity_vector_ use conjugate？
 // Change the 'orientation_' so that it agrees with the current
  // 'gravity_vector_'.
  const Eigen::Quaterniond rotation = Eigen::Quaterniond::FromTwoVectors(
      gravity_vector_, orientation_.conjugate() * Eigen::Vector3d::UnitZ());
orientation_ = (orientation_ * rotation).normalized();
2.what does the commentary mean?
3.why is "orientation_ = orientation_*rotation" not " rotation * orientation_"?

garlinplus, 2019-07-25:
i think it's a bug

sunbingfeng, 2021-01-28:
In my opinion, `gravity_vector_` is with reference to the body frame while `orientation_` means the transformation from body frame to world frame, and `UnitZ` vector stands for the normalized inverse gravity vector in world frame, thus we need the conjugate of `orientation_` to transform it to body frame.