cartographer-project/cartographer #1611: Error linking imu_tracker_test.cc.o: undefined reference to symbol from libgtest.so

lukehutch, 2019-07-12:
I encounter this issue when building cartographer_ros on Fedora 29 (without Ninja):

```
Linking CXX executable cartographer.mapping.imu_tracker_test
/usr/bin/ld: CMakeFiles/cartographer.mapping.imu_tracker_test.dir/cartographer/mapping/imu_tracker_test.cc.o: undefined reference to symbol '_ZN7testing8internal20DoubleNearPredFormatEPKcS2_S2_ddd'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```

Apparently this is because `-lgtest` occurs before `imu_tracker_test.cc.o` on the commandline:

https://stackoverflow.com/a/19905704/3950982

No idea how to fix this with cmake/catkin though...

lukehutch, 2019-07-12:
(this is a non-Ninja build, since Ninja causes different problems...)

It looks like all of `cartographer.common.math_test` is broken:

```
Scanning dependencies of target cartographer.common.math_test
[ 64%] Building CXX object CMakeFiles/cartographer.common.math_test.dir/cartographer/common/math_test.cc.o
[ 64%] Linking CXX executable cartographer.io.proto_stream_test
/usr/bin/ld: CMakeFiles/cartographer.io.proto_stream_test.dir/cartographer/io/proto_stream_test.cc.o: undefined reference to symbol '_ZTIN7testing4TestE'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.io.proto_stream_test.dir/build.make:111: cartographer.io.proto_stream_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:492: CMakeFiles/cartographer.io.proto_stream_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 64%] Linking CXX executable cartographer.common.math_test
/usr/bin/ld: CMakeFiles/cartographer.common.math_test.dir/cartographer/common/math_test.cc.o: undefined reference to symbol '_ZN7testing8internal20DoubleNearPredFormatEPKcS2_S2_ddd'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.common.math_test.dir/build.make:111: cartographer.common.math_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:644: CMakeFiles/cartographer.common.math_test.dir/all] Error 2
[ 65%] Linking CXX executable cartographer.io.proto_stream_deserializer_test
[ 65%] Linking CXX executable cartographer.common.lua_parameter_dictionary_test
/usr/bin/ld: CMakeFiles/cartographer.io.proto_stream_deserializer_test.dir/cartographer/io/proto_stream_deserializer_test.cc.o: undefined reference to symbol '_ZN7testing8internal9DeathTest6CreateEPKcPKNS0_2REES3_iPPS1_'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.io.proto_stream_deserializer_test.dir/build.make:111: cartographer.io.proto_stream_deserializer_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:302: CMakeFiles/cartographer.io.proto_stream_deserializer_test.dir/all] Error 2
/usr/bin/ld: CMakeFiles/cartographer.common.lua_parameter_dictionary_test.dir/cartographer/common/lua_parameter_dictionary_test.cc.o: undefined reference to symbol '_ZN7testing8internal9DeathTest6CreateEPKcPKNS0_2REES3_iPPS1_'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.common.lua_parameter_dictionary_test.dir/build.make:111: cartographer.common.lua_parameter_dictionary_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:454: CMakeFiles/cartographer.common.lua_parameter_dictionary_test.dir/all] Error 2
[ 65%] Linking CXX executable cartographer.mapping.2d.xy_index_test
[ 65%] Linking CXX executable cartographer.sensor.point_cloud_test
/usr/bin/ld: CMakeFiles/cartographer.mapping.2d.xy_index_test.dir/cartographer/mapping/2d/xy_index_test.cc.o: undefined reference to symbol '_ZTIN7testing4TestE'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.mapping.2d.xy_index_test.dir/build.make:111: cartographer.mapping.2d.xy_index_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:188: CMakeFiles/cartographer.mapping.2d.xy_index_test.dir/all] Error 2
/usr/bin/ld: CMakeFiles/cartographer.sensor.point_cloud_test.dir/cartographer/sensor/point_cloud_test.cc.o: undefined reference to symbol '_ZN7testing8internal20DoubleNearPredFormatEPKcS2_S2_ddd'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.sensor.point_cloud_test.dir/build.make:111: cartographer.sensor.point_cloud_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:112: CMakeFiles/cartographer.sensor.point_cloud_test.dir/all] Error 2
[ 66%] Linking CXX executable cartographer.mapping.3d.submap_3d_test
[ 66%] Linking CXX executable cartographer.mapping.2d.probability_grid_test
/usr/bin/ld: CMakeFiles/cartographer.mapping.3d.submap_3d_test.dir/cartographer/mapping/3d/submap_3d_test.cc.o: undefined reference to symbol '_ZN7testing8internal20DoubleNearPredFormatEPKcS2_S2_ddd'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.mapping.3d.submap_3d_test.dir/build.make:111: cartographer.mapping.3d.submap_3d_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: CMakeFiles/cartographer.mapping.3d.submap_3d_test.dir/all] Error 2
[ 66%] Linking CXX executable cartographer.mapping.2d.submap_2d_test
/usr/bin/ld: CMakeFiles/cartographer.mapping.2d.probability_grid_test.dir/cartographer/mapping/2d/probability_grid_test.cc.o: undefined reference to symbol '_ZN7testing8internal20DoubleNearPredFormatEPKcS2_S2_ddd'
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.mapping.2d.probability_grid_test.dir/build.make:111: cartographer.mapping.2d.probability_grid_test] Error 1
[... cut ...]
/usr/bin/ld: //usr/lib64/libgtest.so.1.8.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cartographer.mapping.imu_tracker_test.dir/build.make:111: cartographer.mapping.imu_tracker_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:340: CMakeFiles/cartographer.mapping.imu_tracker_test.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'cartographer': 
  Command '['/home/luke/catkin_ws/install_isolated/env.sh', 'make', '-j16', '-l16']' returned non-zero exit status 2
```

lukehutch, 2019-07-12:
The problem is due to incompatibility between Cartographer and the system packages `gflags`, `glog`, `gmock`, `gtest`.

I had to remove these packages, and compile and install all these packages from source to get most of cartographer_ros to build. However, cartographer itself still does not build, even with these freshly-built libraries:

```
[236/334] Linking CXX executable cartographer.mapping.internal.2d.pose_graph_2d_test
FAILED: cartographer.mapping.internal.2d.pose_graph_2d_test 
: && /usr/bin/c++  -O2 -DNDEBUG  -rdynamic CMakeFiles/cartographer.mapping.internal.2d.pose_graph_2d_test.dir/cartographer/mapping/internal/2d/pose_graph_2d_test.cc.o  -o cartographer.mapping.internal.2d.pose_graph_2d_test  libcartographer.a -lgmock_main -lpthread libcartographer_test_library.a libcartographer.a /home/luke/catkin_ws/install_isolated/lib64/libceres.a /usr/lib64/libglog.so.0.4.0 /usr/lib/libgflags.so.2.2.2 -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lmetis -lcxsparse -llapack -lblas -lsuitesparseconfig -lrt -lmetis -lcxsparse -lgomp -llua-5.3 -lm -lboost_iostreams -lboost_regex -lglog /usr/lib/libgflags.so.2.2.2 -lpthread -lcairo -lprotobuf -lpthread && :
/usr/bin/ld: CMakeFiles/cartographer.mapping.internal.2d.pose_graph_2d_test.dir/cartographer/mapping/internal/2d/pose_graph_2d_test.cc.o: undefined reference to symbol '_ZN7testing8internal24FormatMatcherDescriptionEbPKcRKSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS9_EE'
/usr/bin/ld: //usr/lib64/libgmock.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[237/334] Linking CXX executable cartographer.sensor.point_cloud_test
FAILED: cartographer.sensor.point_cloud_test 
: && /usr/bin/c++  -O2 -DNDEBUG  -rdynamic CMakeFiles/cartographer.sensor.point_cloud_test.dir/cartographer/sensor/point_cloud_test.cc.o  -o cartographer.sensor.point_cloud_test  libcartographer.a -lgmock_main -lpthread libcartographer_test_library.a libcartographer.a /home/luke/catkin_ws/install_isolated/lib64/libceres.a /usr/lib64/libglog.so.0.4.0 /usr/lib/libgflags.so.2.2.2 -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lmetis -lcxsparse -llapack -lblas -lsuitesparseconfig -lrt -lmetis -lcxsparse -lgomp -llua-5.3 -lm -lboost_iostreams -lboost_regex -lglog /usr/lib/libgflags.so.2.2.2 -lpthread -lcairo -lprotobuf -lpthread && :
/usr/bin/ld: CMakeFiles/cartographer.sensor.point_cloud_test.dir/cartographer/sensor/point_cloud_test.cc.o: undefined reference to symbol '_ZN7testing4TestC2Ev'
/usr/bin/ld: //usr/lib64/libgtest.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[251/334] Building CXX object CMakeFiles/cartographer.mapping.pose_extrapolator_test.dir/cartographer/mapping/pose_extrapolator_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/luke/catkin_ws/install_isolated/env.sh', 'ninja', '-j16', '-l16']' returned non-zero exit status 1
```

kunaltyagi, 2019-12-02:
I resolved the issue by modifying the `FindGMock.cmake`. Maybe someone else can propose a more elegant solution. Else I can propose a PR with the following diff

```diff
diff --git a/cmake/modules/FindGMock.cmake b/cmake/modules/FindGMock.cmake
index c663e98..9afa068 100644
--- a/cmake/modules/FindGMock.cmake
+++ b/cmake/modules/FindGMock.cmake
@@ -30,6 +30,14 @@ if(NOT GMock_FOUND)
     PATHS
       /usr
   )
+  find_library(GTEST_LIBRARIES
+    NAMES gtest
+    PATH_SUFFIXES lib
+  )
+  find_library(GMOCK_A_LIBRARIES
+    NAMES gmock
+    PATH_SUFFIXES lib
+  )
 
   # Find system-wide gtest header.
   find_path(GTEST_INCLUDE_DIRS gtest/gtest.h
@@ -65,7 +73,7 @@ if(NOT GMock_FOUND)
 
   # System-wide installed gmock library might require pthreads.
   find_package(Threads REQUIRED)
-  list(APPEND GMOCK_LIBRARIES ${CMAKE_THREAD_LIBS_INIT})
+  list(APPEND GMOCK_LIBRARIES ${GMOCK_A_LIBRARIES} ${GTEST_LIBRARIES} ${CMAKE_THREAD_LIBS_INIT})
 endif()
 
 include(FindPackageHandleStandardArgs)
```

csorvagep, 2019-12-02:
@kunaltyagi Thanks, it worked for me.

cottsay, 2020-07-29:
@kunaltyagi, your patch worked for me on Fedora 32. It would be great to see this merged.

kunaltyagi, 2020-07-29:
Did you test with the master? There appears to be a similar code update on 2020 June 10, commit db85e089

cottsay, 2020-07-29:
I'm using the released version of cartographer for Eloquent, not master.

That change appears to affect the include directories, but this issue describes a linking problem. I tried applying the commit you referenced, but it does not resolve this issue.

meyerj, 2021-03-17:
I just encountered a similar linking issue in a Ubuntu 18.04 environment, with system packages `libgtest-dev` and `google-mock` installed, and with compile flag `-Wl,--no-undefined`.

The problem is that [FindGMock.cmake](https://github.com/cartographer-project/cartographer/blob/105c034577220268cd28a304a185adbec46b729f/cmake/modules/FindGMock.cmake#L79) sets variable `GMOCK_LIBRARIES`, but in [CMakeLists.txt:313](https://github.com/cartographer-project/cartographer/blob/105c034577220268cd28a304a185adbec46b729f/CMakeLists.txt#L313) the `${TEST_LIB}` (i.e. `cartographer_test_library`) target is only linked against `${GMOCK_LIBRARY}`, which is unset. Although all test targets are linked to `${GMOCK_LIBRARIES}` via the `google_test()` CMake function defined in [cmake/functions.cmake](https://github.com/cartographer-project/cartographer/blob/9675d63926282f505f2e8333dd919429e49862c9/cmake/functions.cmake#L42), there might be linking order issues if the `${TEST_LIB}` itself is not linked against gtest and gmock.

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index fe1289d..27e4aac 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -301,7 +301,7 @@ target_include_directories(${TEST_LIB} SYSTEM PRIVATE
 if (WIN32)
   target_compile_definitions(${TEST_LIB} PUBLIC -DGTEST_LINKED_AS_SHARED_LIBRARY)
 endif()
-target_link_libraries(${TEST_LIB} PUBLIC ${GMOCK_LIBRARY})
+target_link_libraries(${TEST_LIB} PUBLIC ${GMOCK_LIBRARIES})
 target_link_libraries(${TEST_LIB} PUBLIC ${PROJECT_NAME})
 set_target_properties(${TEST_LIB} PROPERTIES
   COMPILE_FLAGS ${TARGET_COMPILE_FLAGS})
```