cartographer-project/cartographer #1613: The issue of Remaining work items in queue

huangshuqing, 2019-07-17:
Hello 
when  I run cartographer with pure Localization in a Given Map, I always  meet the issue of remaining work items in queue，such as  "Remaining work items in queue: 17893". The size of queue frequently reaches tens of thousands. From my personal point of view, If the parameters are tuned well, the value should be 0. However, How can I continue to tune the parameters, can you give me some advice?

huangshuqing, 2019-07-17:
@SirVer  @damienrg @wjwwood @damonkohler

Yu-Xiaoxian, 2019-07-25:
This tips in cartographer_ros about low lantency of Global SLAM may helped. 
[Tuning methodology](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency)

huangshuqing, 2019-07-26:
thanks, I have solved it.

Yu-Xiaoxian, 2019-08-07:
How did the problem get solved, would you like to share your experience?

mojovski, 2019-09-04:
yeah, how did you solve it? I have a similar issue.
in general: What is a work item and what can help to increase its processing speed?
* increase number of background threads?
* ...? Running out of ideas (dont want to reduce other "quality" parameters inside the lua files)

MichaelGrupp, 2020-10-17:
Work items are for example constraint searches and optimization steps. Following the tuning guide can help - e.g. if you decrease the constraint builder sampling ratio there will be less constraint searches enqueued (which are a costly operation and usually shouldn't be done for every node, thus sampling).

JaredLee1218, 2021-04-01:
@huangshuqing 
Hi, would you like to share your result Please..T ^ T