cartographer-project/cartographer #1617: Unable to achieve global loop closure between sub-maps

fayyazpocker, 2019-07-26:
Hello guys,

I am trying to generate a map of an industry. I am not able to achieve a proper loop closure as shown below:
[image]

Inter constraints between submaps:
[image]

The two distinct lines you see in the image below should have been collinear:
[image]

You can find the bag file related to the above image in  [drive](https://drive.google.com/drive/folders/1xG1Nh55VlaH01cf0gWSneZqTRijmxmhY?usp=sharing) named  "cartographer_issue.bag". You can also find another rosbag named "complete_run.bag" which also have the same topics but with a longer time.

Please find the [configuration](https://drive.google.com/file/d/1DOaYXwGYR_z6TpF5VNK1ZHGQ_TDmjZnE/view?usp=sharing) i used for running cartographer

According to #141 and #127, i tried changing huber scale, number of scans per submap and minimum score required for loop closure detection accordingly but still am not able to get an appreciable map.

fayyazpocker, 2019-07-27:
@SirVer @damienrg @wjwwood @damonkohler

Yu-Xiaoxian, 2019-08-08:
Did your problem get solved?

fayyazpocker, 2019-08-19:
Thanks for replying. I tried to tune the following parameters for local slam keeping the sampling ratio as zero: 
- TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight
- TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight
- TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability
- TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability

Then after getting appreciable results, i tuned for global slam by tuning the following parameters:
- POSE_GRAPH.constraint_builder.min_score
- POSE_GRAPH.constraint_builder.sampling_ratio
- POSE_GRAPH.optimization_problem.huber_scale

It was a tedious process to tune these parameters as i was not able to understand the pattern of the parameters. So i had to make a bash script to run different parameters and take screenshots of the RViz for two days and finally ended up with this configuration:

TRAJECTORY_BUILDER_2D.min_range = 0.05 
TRAJECTORY_BUILDER_2D.max_range = 8
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = false
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight=1200
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight=600
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 100
POSE_GRAPH.optimize_every_n_nodes = 100
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.hit_probability = 0.65
TRAJECTORY_BUILDER_2D.submaps.range_data_inserter.probability_grid_range_data_inserter.miss_probability = 0.40
POSE_GRAPH.constraint_builder.min_score=0.75
POSE_GRAPH.optimization_problem.huber_scale=1
POSE_GRAPH.constraint_builder.sampling_ratio=0.3

This configuration gave me a good result as shown below:
[image]

I am not sure if this is the correct way to tune the paramters by trial and error. It would be helpful if you could suggest a more systematic way to tune the parameters. Also, if possible can you please have a look at the bag file and the new configuration?