cartographer-project/cartographer #1623: What is the relationship between the value of submaps.high_resolution and robot positioning accuracy?

huangshuqing, 2019-08-13:
hello,
I am tuning the parameter of cartographer according to https://google-cartographer-ros.readthedocs.io/en/latest/#. I want to know What the relationship is between the value of submaps.high_resolution and robot positioning accuracy? 
thanks for your answer.

Yu-Xiaoxian, 2019-09-11:
According to may knowledge, cartographer provide sub-pixel accuracy using ceres-scan-match. But high resolution does not always provides high positioning accuracy, as ceres-scan-match is very sensitive to initial pose and has rather small convergence domain(which means the result of scan-match is likely to diverge).

qmx066, 2019-09-19:
if everything is fine, resolution size impacts accuracy, the smaller the better (optimization is converged).