cartographer-project/cartographer #1625: Why the score of unknown area is highest?

yuzy2019, 2019-08-19:
In fast correlative scan matcher, the score of unknown area is highest, even exceed the maximum. In [fast_correlative_scan_matcher.cc](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/2d/scan_matching/fast_correlative_scan_matcher_2d.cc) , line 111, the cost of grid is calculated as follow
`
1.f - std::abs(grid.GetCorrespondenceCost(Eigen::Array2i(0, y))))
`
while the CorrespondenceCost of grid is from 0.1 to 0.9, 0 for unknown area, which defined in [probability_values.h](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/probability_values.h). As a result, the unknown area get the highest score 1 for scan matching, am I right? Will this result in misleading in scan matching?

kdaun, 2019-08-19:
The correspondence cost was introduced as a unified measure relating how well an observation at a grid point corresponds with the environment model. Thereby, enabling a common FastCorrelativeScanMatcher for TSDFs and probability grids.
As defined in [probability_values.h](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/probability_values.h#L60) holds:
`correspondence cost = 1 - probability value`
Therefore, unknown areas are assigned a low score.

yuzy2019, 2019-08-19:
@kdaun I see, but I think the probability values for unknown area is 0.f, as defined in [probability_values.h](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/probability_values.h#L80)
`
constexpr uint16 kUnknownProbabilityValue = 0
`
As a result, the correspondence_cost will be 1.0f, even higher than the max probability.

kdaun, 2019-08-19:
> As a result, the correspondence_cost will be 1.0f, even higher than the max probability.

Yes, this is the intended behavior. A high correspondence cost indicates a poor match and yields a low matching score (see [fast_correlative_scan_matcher_2d Line 111](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/internal/2d/scan_matching/fast_correlative_scan_matcher_2d.cc#L111)).

yuzy2019, 2019-08-20:
@kdaun, thanks for your help, finally I have it figured out! Below is my understanding

Actually there are four different values and I have them mixed up. These four values are: correspondence_cost, correspondence_cost_value, probability, probability_value, and scan matcher use correspondence_cost. As defined in [probability_values.cc Line 62](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping/probability_values.cc#L62), 0.9 is for free and unknown, 0.1 is for occupied.

I have draw a diagram to show their relationships.(see [map_description](https://github.com/yuzy2019/learnCartographer/blob/master/probability_correspondence_cost.png))

closed 2019-08-20 by yuzy2019